jotego/jtcores #678: 1942: graphic bug with Boss (big airplane)

Lmn-San, 2024-05-26:
Every 8 levels there is a boss appearing. With vertical lines missing.
[image]

jotego, 2024-05-27:
Could you please share a snapshot of your DIP settings?

Lmn-San, 2024-05-27:
Here it is. (Playing on MiSTer)
[image]

jotego, 2024-05-27:
Well, believe it or not, this is a genuine bug but it only happens with that flip setting. Toggle the flip setting, reset the machine and it will be gone.

It is the same issue that also causes this screen if you wait long enough in attract mode:

[image]

The CAPCOM logo appears split. 

I could fix it, but the real hardware was shown to behave this way, so I'm leaving it. The thing we can do is to have the good flip setting as default. If you play through HDMI, rotating the video, there is no penalty in turning flip on.

closed 2024-05-27 by jotego

Lmn-San, 2024-05-28:
Thank you. I didn't knew that was a genuine bug.
By turning the flip on it worked as you mentioned.