jotego/jtcores #686: s18: bad graphics

jotego, 2024-06-03:
These games have bad GFX. The tile bank seems wrong (a8cad670).

- [ ] hamaway
- [ ] cltchitr

jotego, 2024-06-09:
Enemies are not visible in ddcrew. It might be related to the tile map.

gyurco, 2024-06-10:
Here there are more inputs than sa, sb and fix from the tilemap chip:
[image]

Just for record, the logic:
https://wiki.pldarchive.co.uk/index.php?title=315-5373

gyurco, 2024-06-10:
Do you have an idea what are the Tilemap0-4 lines on the tilemap chip?
Also objcol[1:0] is used here for vdp/s16 priority, but in s16 itself, obj_prio=obj_pxl[11:10], strange.

jotego, 2024-06-12:
I am not sure. The arcade GFX chipset seems to be a redesign of the System 16B chipset with slightly different IO. The original chipset seemed to identify which tilemap layer was active. The priority of that layer may have been available too. The file s16b/doc/s16b.txt has a good and short description of the layer priorities around line 1086.

Because S16B video comes already mixed, S18 cannot do much when merging the VDP video. Particularly, you cannot have sprites hide each other depending on the y position. That's why the sprites used on the VDP look flat in Alien Storm, so as to make this limitation less noticeable.

MAME identifies some scenes and priority values but they ignore the real connections on the board. I have been trying to make sense of the decoding signals in the PAL (315-5373) but I could only be sure of what I wrote in the colmix _case_ statement.

gyurco, 2024-06-12:
Shouldn't this VDP-S16 priority thing belong to a different issue?

jotego, 2024-06-12:
That's fine. Let's limit this one to fix the garbled graphics. Layer priority will be handled on a different one.

gyurco, 2024-06-12:
I opened https://github.com/jotego/jtcores/issues/701

jotego, 2024-06-14:
## Fixed

Clitch Hitter seems ready to go. Thank you.
**hamaway** and **desertbr* have a layer priority problem (#701) but tile mapping seems fine.
For **lghost** I cannot pass the calibration screen so I do not know yet.

## ddcrew

This is the image that should appear right after booting up ddcrew. The tilemap bank is pointing to a different place, producing a completely different scene.

[image]

## wwallyj

The tile bank is wrong in several scenes, including the title

[image]

terminator2k2, 2024-06-14:
@jotego i have a version of laser ghost which is modded and uses joystick controls and im sure it bypasses the calibration screen. I also have a  wally modded rom which uses joystick controls aswell. Theres also Clockwork Aquario.

jotego, 2024-07-04:
> @jotego i have a version of laser ghost which is modded and uses joystick controls and im sure it bypasses the calibration screen. I also have a wally modded rom which uses joystick controls aswell. Theres also Clockwork Aquario.

It looks like these ROM sets have not been documented. It would be good to report it to the MAME team so they set a name for the files and a CRC check sum for preservation.

jotego, 2024-07-13:
ddcrew seems fine in 8533440b after the priority edits done for mwalk.

gyurco, 2024-07-13:
For me, it looks like desertbr also good. And wwally, too, but I have a problem adding trackball controls. The trackball input is using an upd4701. As there's jt4701, it seemed straightforward. Unfortunately dial_x and dial_y (which jt4701 accepts) don't correspond to mouse x and y, but only mouse x. Is it feasible to add the vertical mouse movement as dial_y emulation?

gyurco, 2024-07-13:
Another issue that the game seems to need reversed dials, but
```
dial = [
    { machine="wwallyj", reverse=true },
]
```
doesn't do anything, MRA has still:
```
    <rom index="1">
        <part>00 6E</part>
    </rom>
```

gyurco, 2024-07-13:
But the later one is not really important, it's easy to reverse the dial signals.
Here's a patch for the controls:
https://github.com/jotego/jtcores/commit/7e14dcf8a103e6c63e4106a1e93c5e069509a0fd
Looks like there's still a minor problem with the layer priority, as the mouse cursor tends to disappear behind the scenes.

jtmiki, 2024-07-24:
It looks like this issue is completed and the rest of the gfx problems are priority-layer related, and addresses in #701. Closing this one.