jotego/jtcores #691: shouse: dspirit does not react to changes to DIP settings

jotego, 2024-06-03:
It was reported in Discord that changing the flip setting in Dragon Spirit does not work. @jtmiki, could you test it?

jtmiki, 2024-06-03:
The game doesn't work when you try to change "Scandoubler Fx" in the Video setting.
Commit: a8cad67

jotego, 2024-06-26:
The flip setting is changed in the service menu. You enter the service menu using the DIP switches. Changing it does change the **dip_flip** signal in the game module. That affects MiSTer's HDMI rotation settings so it flips the screen but the tilemap seems to be flipped by the software at the same time. So the net effect is no flip in the CRT, while there is flip in the HDMI output because of MiSTer's rotation setting changing.

jotego, 2024-06-26:
The flip mode does not work at all. There is no way to flip the tilemap at the moment, even if the flip signal going into the tilemap module is altered manually.

gyurco, 2024-08-02:
Where's the flip setting?

The hardware has a software flip register, some games set it, others not. But it cannot be override externally, as it requires some cooperation from the software side (coordinates are different in the two modes).

jotego, 2024-08-02:
Use the DIP settings to enter the service mode in _Dragon Spirit_. You will see options that presumably get saved in the EEPROM. These two screenshots are from MAME:

Default:
[image]

And when you turn it on:
[image]

gyurco, 2024-08-02:
I see now.
The current code is aware of the changed coordinates in flip/non-flip modes, but that's all, it doesn't actually draws a flipped screen.

gyurco, 2024-08-03:
Just a side question: is 'rotate controls' broke recently? The OSD option is there, but it doesn't do anything even if I switch between 'yes' and 'no'. I also noticed it in kchamp.

jotego, 2024-08-04:
I do not remember doing anything to it lately...

gyurco, 2024-08-04:
> I do not remember doing anything to it lately...

The problem was that MRA tool didn't handle rom index=1 when it was more than one byte. I updated the tool.

jotego, 2024-08-05:
I had to override core_mod calculation for this core as the information coming from dip_flip contradicts MAME's database for Pac-Mania. See f548a46

This task is completed. Thanks @gyurco

closed 2024-08-05 by jotego

gyurco, 2024-08-05:
> I had to override core_mod calculation for this core as the information coming from dip_flip contradicts MAME's database for Pac-Mania. See [f548a46](https://github.com/jotego/jtcores/commit/f548a46857db72a126da51f77f6dd74b6884c470)
> 

As I see, Galaga88 also flipped by default.

stiegis, 2024-08-07:
Is this being realeased in the next update? Cannot flip cores on a crt monitor but works in hdmi perfectly. No big problem as I can swap joke connectors on the monitor but this is not very practical as other games cannot be flipped then

jotego, 2024-08-07:
> Is this being realeased in the next update? Cannot flip cores on a crt monitor but works in hdmi perfectly. No big problem as I can swap joke connectors on the monitor but this is not very practical as other games cannot be flipped then

Yes, it will be released in the next update.

stiegis, 2024-08-07:
thanks Jotego thats great