jotego/jtcores #692: s18: try different length for HS/VS signals

jotego, 2024-06-03:
Some monitors are troubled with JTS18 analogue video output. Try this:

- [ ] Check that HS length is like the original (VS already checked). The original screenshot is in the doc folder.
- [ ] Compile the core for longer HS lengths and make the file available here for users to check

PapiFed, 2024-06-03:
Just FYI using DV I have:
No sync on a Sony PVM-2950
Sync on a JVC TMH-150cg

rp-jt, 2024-06-05:
I have compiled the core for varying manually the length of HS and VS
You can find the rbf file here:

[jts18.rbf.zip](https://github.com/user-attachments/files/15592645/jts18.rbf.zip)

To do so, please press the keys [+] or [-] (or equivalent to where they are in the Spanish keyboard) to adjust horizontally and "Shift + [+]" or "Shift +  [-]" to adjust vertically. I leave the position of the keys in the Spanish keyboard below for reference.

To go back to the original settings, Press either "Ctrl + [+]" or "Ctrl + [-]"

An eight-digit number should come up in the bottom of the screen when doing so. The four digits on the right are associated with HS and the 4 in the left with VS. Please let us know if there's an eight digit combination that works for you.
Thanks

[image]

Schermobianco, 2024-06-05:
My test on PVM Panasonic BT-1415DA

https://streamable.com/g5cjcy

PapiFed, 2024-06-06:
Are Sync width and sync length the same thing ?

jotego, 2024-06-06:
> Are Sync width and sync length the same thing ?

Yes. The important parameters are the synchronization signal frequency and its active time (expressed as pulse width or length)

PapiFed, 2024-06-06:
> > Are Sync width and sync length the same thing ?
> 
> Yes. The important parameters are the synchronization signal frequency and its active time (expressed as pulse width or length)

Having the option to modulate the vertical sync width in all Cores would be a dream for all high end Ikegami monitors owners (probably JVC DT-V owners also). It is currently only available in a few Cores.

Fpg-Lee, 2024-06-06:
> Played around quite a bit with this test Core but could not get any non-scrambled image to display on the 2950.

Same here, implementation would benefit from H+V adjustments within the visible GUI via HDMI to stabilize CRT syncs.
P.s the controls wouldn't respond after config for me, I couldn't get a coin to insert to further test game inputs on either game.

Fpg-Lee, 2024-06-06:
'Dirty' S18 core control pads don't respond after before/after config for me, I couldn't get a coin to insert to further test game inputs on either game. Thanks

deepthaw, 2024-06-06:
FYI ] and / are the US Keyboard equivalents of + and /

deepthaw, 2024-06-06:
Went through all 255 possibilities (I assume only the top octet should change?) and did not find anything that synced. Values ending in 2 seemed to be the closest, with no rolling but still clearly off.

Fpg-Lee, 2024-06-06:
> FYI ] and / are the US Keyboard equivalents of + and /

Thanks. Have you tried the core via HDMI? the Core will not add coins, the BETA does.

deepthaw, 2024-06-06:
> > FYI ] and / are the US Keyboard equivalents of + and /
> 
> Thanks. Have you tried the core via HDMI? the Core will not add coins, the BETA does.

I did but I didn't even try to insert coins. I spent the whole time stepping through the sync values.

terminator2k2, 2024-06-07:
> > > FYI ] and / are the US Keyboard equivalents of + and /
> > 
> > 
> > Thanks. Have you tried the core via HDMI? the Core will not add coins, the BETA does.
> 
> I did but I didn't even try to insert coins. I spent the whole time stepping through the sync values.

its likely a debug core , made with no inputs/sound

jotego, 2024-06-08:
So far three users tried this debug build and it only helped one (@Schermobianco). @Schermobianco's video is not available anymore but I think the result was that going from a zero to a non-zero value made it visible. I measured the PCB and found that HS was 6% longer there (equivalent to **2 pixels**). I have modified the core to match the PCB length (b7015c27). The extra 320ns may help some users but from the reports here it looks like this change will benefit few people.

HS signal measured on the board: 5.32us active time length
[image]

Schermobianco, 2024-06-08:
Hi @jotego , no it doesn't help me the better sync I had was the same as the original core (top left sync issue)

[image]

Fpg-Lee, 2024-06-08:
> Hi @jotego , no it doesn't help me the better sync I had was the same as the original core (top left sync issue)
> 
> ![IMG_20240608_081913.jpg](https://private-user-images.githubusercontent.com/89546473/337854953-0197067d-c27b-41b0-b2ec-32d921ee06b0.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTc4Mjc5OTEsIm5iZiI6MTcxNzgyNzY5MSwicGF0aCI6Ii84OTU0NjQ3My8zMzc4NTQ5NTMtMDE5NzA2N2QtYzI3Yi00MWIwLWIyZWMtMzJkOTIxZWUwNmIwLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA2MDglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNjA4VDA2MjEzMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTBiNmIzNzI4YzUzOTJhMmYwMzc0ZTA4NjljYjRjYzNkOGY3ZDhhMzZiMDI5ZDQxNTAxOGUwY2ZiNWNhYTgxZTMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.w37N5JXkHnTF3NL8D_OUFIE-zC4bn98Zs40YOo5OiAM)

Is the updated Core compiled and available? Thanks.

Schermobianco, 2024-06-08:
> Is the updated Core compiled and available? Thanks.

My photo is from the test core, the updated core compiled isn't available right now

Fpg-Lee, 2024-06-08:
> My photo is from the test core, the updated core compiled isn't available right now

Gotcha, I ran the updater incase Jose added it to Update all, I got my fingers crossed it improves CRT sync compatibility for many CRT users.

bazset, 2024-06-08:
I tried the first custom core posted. I tried every combination available but nothing synced the picture (Sony PVM).

@jose  when will the new core be posted? will it be combined into the update_all core or will it need to be manually downloaded?

jotego, 2024-06-08:
> @jotego when will the new core be posted? will it be combined into the update_all core or will it need to be manually downloaded?

[jts18_8db7b95.zip](https://github.com/user-attachments/files/15748391/jts18_8db7b95.zip)

This version has exactly the same HS/VS signals as the original board. Let me know if it works.

Schermobianco, 2024-06-08:
> [jts18_8db7b95.zip](https://github.com/user-attachments/files/15748391/jts18_8db7b95.zip)
>
> This version has exactly the same HS/VS signals as the original board. Let me know if it works.

@jotego Same here

https://github.com/jotego/jtcores/assets/89546473/009137fc-dadd-40da-9f73-c4aefa6490c8

jotego, 2024-06-08:
Thanks for testing it. We will build another debug version just focusing on HS as it looks like the vertical signal gets locked correctly.

bazset, 2024-06-09:
@jotego  I tried the new core in jts18_8db7b95.zip and this is what i got. same as the original core. 

https://github.com/jotego/jtcores/assets/78243398/36ff76a6-3374-46ae-9736-a188a9c5b79d

rogersouza-work, 2024-06-10:
> @jotego I tried the new core in jts18_8db7b95.zip and this is what i got. same as the original core.
> 
>  sd.mp4

I've got the same image on my arcade (Toshiba tri-sync monitor)

danieldoyle, 2024-06-10:
> > @jotego I tried the new core in jts18_8db7b95.zip and this is what i got. same as the original core.
> > sd.mp4
> 
> I've got the same image on my arcade (Toshiba tri-sync monitor)

Which model?  Most arcade monitors have hfreq/hhold adjustments that will solve it

rogersouza-work, 2024-06-10:
> > > @jotego I tried the new core in jts18_8db7b95.zip and this is what i got. same as the original core.
> > > sd.mp4
> > 
> > 
> > I've got the same image on my arcade (Toshiba tri-sync monitor)
> 
> Which model? Most arcade monitors have hfreq/hhold adjustments that will solve it

That's not the problem. I've got some arcade PCBs, Mister Fpga, Raspberry Pi and It only happens with this core.

rogersouza-work, 2024-06-10:
> > > @jotego I tried the new core in jts18_8db7b95.zip and this is what i got. same as the original core.
> > > sd.mp4
> > 
> > 
> > I've got the same image on my arcade (Toshiba tri-sync monitor)
> 
> Which model? Most arcade monitors have hfreq/hhold adjustments that will solve it

My chassis/monitor is a Toshiba PF D29C051

jotego, 2024-06-10:
Thank you for all the feedback.

The monitor model is actually important. This does not happen to everyone. It does not happen on any of the screens we can test it. The core has the **same video signals as the original board**, so I would assume that most arcade monitors should work well with it. We'll check if we can buy one of these screens for an affordable price as I am always up for making the analog video output more compatible.

In the meantime, as I said above, we will prepare another debug core allowing other settings and upload it here.

bazset, 2024-06-10:
my monitor is a Sony PVM-1953MD

Schermobianco, 2024-06-10:
PVM Panasonic BT-S1415DA

Fpg-Lee, 2024-06-10:
My consumer CRT=    iLO DTV2794

jotego, 2024-06-14:
We are looking into adding more advance HS/VS customization options in the OSD menu, but it will still not be enough for all monitors and cores. Some monitors just not lock certain frequencies in time to display the active video part of the signal.

I think replacing the capacitors on these monitors may have a chance of improving the PLL performance. Particularly, around the supply decoupling and the horizontal synchronization circuit. Please consider servicing your monitors.

rogersouza-work, 2024-06-14:
I have a System 18 Shadow Dancer original PCB and I have no problems with sync in both Egret 3 or Sega Blast City. 
I am sorry but I can not agree with your argument. Lots of people are having sync problems ONLY with this core. How can this be a recap problem?

jotego, 2024-06-14:
The core produces the same frequency and width for all the sync pulses. But there is one difference I cannot control from the core: **the voltage**. That will also have an effect on the PVM PLL. The board does not have the same lov/high level voltages as a MiSTer FPGA. These voltage levels are fixed and cannot be modified by the core.

Replacing the capacitors may help because PLL's have a low-pass filter in the loop that will be dominated by some capacitor, which might be discrete in these monitors. Reducing supply noise will also help in having the PLL lock up more quickly.

> Lots of people are having sync problems

I only count the users in this thread. Even if only a few users, we have been searching for the PVM models that fail to buy one. We have not found one yet. Our test equipment works correctly with this core and we cannot reproduce the issue.

The only thing we can do is give a finer control on the sync signals to the user and let you play with different combinations until one works. Another possibility is to enable a higher sync rate (closer to 60Hz), which should help too at the cost of increasing game speed. I am not a big fan of this second option.

jotego, 2024-06-14:
The blue line is an original SYSTEM 18 PCB, the yellow one is the core on a SiDi128. As you see the voltage levels are different but the frequency and width are equal. Everyone, how do you produce the video signal? Is it direct video (HDMI to VGA converter), the old analog IO board or the new digital IO board?

[image]

rogersouza-work, 2024-06-14:
> The blue line is an original SYSTEM 18 PCB, the yellow one is the core on a SiDi128. As you see the voltage levels are different but the frequency and width are equal. Everyone, how do you produce the video signal? Is it direct video (HDMI to VGA converter), the old analog IO board or the new digital IO board?
> 
> ![DS1Z_QuickPrint4](https://private-user-images.githubusercontent.com/1863036/339764501-edca5e45-e251-4092-9be6-bd8d1a4ad4b5.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTgzNzAzNzMsIm5iZiI6MTcxODM3MDA3MywicGF0aCI6Ii8xODYzMDM2LzMzOTc2NDUwMS1lZGNhNWU0NS1lMjUxLTQwOTItOWJlNi1iZDhkMWE0YWQ0YjUucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDYxNCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA2MTRUMTMwMTEzWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YjFkZmMzMDZkZDRmNWVkYjg0MTg5ZmVjMDRjN2IxOTZhZTE3ZGMwYTRlMDE1YWQ3MGI0NjA2NTYwMDQyZWU1MyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.bkhwaIOhnPX1sDZNoOEw7dpozqaj3Tn0LayvnJ2R5mI)

I am using an analogue I/O board with Antonio's Villena Jamma Adapter. 
Anyway, I am having the same sync problem using analogue I/O straight on the JVC TM-H1750CG

rogersouza-work, 2024-06-14:
> I am using an analogue I/O board with Antonio's Villena Jamma Adapter. Anyway, I am having the same sync problem using analogue I/O straight on the JVC TM-H1750CG

[image]

danieldoyle, 2024-06-14:
For anyone on this thread feeling really confident caps don’t play a role in sync:

https://youtu.be/oMQq2As55Ho?si=ltAViKN2PGNzR7bj

bazset, 2024-06-14:
I am using the VGA out on a MiSTer Mini-ITX Ironclad Plus board.

HawkeyeRetro, 2024-06-14:
Jotego asked me to share my sync issue in this thread. 

I have a 6.0 I/O board and have noticed issues with sync for the TMNT, Simpsons, and Splatter House cores. I have no other sync issues with any other cores for arcades or consoles.

My output is via component cables. I have tried these cores on two different monitors with the exact same result. One is a Toshiba A model (consumer CRT) the other is a Sony PVM 14m4u. 

I find it perplexing that the sync issue manifests exactly the same on two different monitors. Thanks! 
[image]

https://github.com/jotego/jtcores/assets/172640919/a185afdf-8ca3-4a5c-908d-d41404308038

https://github.com/jotego/jtcores/assets/172640919/020b64f2-5ec3-4b0d-9dad-6b04ac966997

[image]

PapiFed, 2024-06-14:
Using Direct Video through an HDFury3
Syncs on a JVC TM-H150cg with a clone input card.
No sync on a Sony PVM-2950

PapiFed, 2024-06-14:
> For anyone on this thread feeling really confident caps don’t play a role in sync:
> 
> https://youtu.be/oMQq2As55Ho?si=ltAViKN2PGNzR7bj
> 
> just to clarify what’s happening in the video if it isn’t obvious:
> 
> 1. No sync on cold start
> 2. Caps get toasty gradual sync happens
> 3. Full synched, power off and on and out of sync
> 4. Repeat this a number of times and the window of time decreases as the caps are already warmed up
> 
> Then wait a few minutes and we get back to cold start state. Mister can be left on of course. So yeah this TV happens to be right on the edge so it’s a good demo of why caps matter.

How do you know it is the caps and not the flyback ?

jotego, 2024-06-14:
> How do you know it is the caps and not the flyback ?

The flyback generates high voltage to drive the tube. It is not related to sync signal processing. Having said that, if the caps in the flyback are degraded, it will generate more switching noise. This noise may couple into the PLL supply and affect the PLL lock performance. That's why replacing decoupling caps may also help.

rogersouza-work, 2024-06-14:
https://github.com/jotego/jtcores/assets/169895421/b387395d-b225-473e-a45c-c8f8697c8f9e

terminator2k2, 2024-06-14:
i use direct video and have no problems when used with my Sony consumer crt tv

jotego, 2024-06-15:
Let's try a shorter HS than the original. This compilation uses 3us for HS width, which is a more typical value. Let me know how it goes.

[jts18_6ae69c20.rbf.zip](https://github.com/user-attachments/files/15845819/jts18_6ae69c20.rbf.zip)

Schermobianco, 2024-06-15:
> Let's try a shorter HS than the original. This compilation uses 3us for HS width, which is a more typical value. Let me know how it goes.
> 
> [jts18_6ae69c20.rbf.zip](https://github.com/user-attachments/files/15845819/jts18_6ae69c20.rbf.zip)
> 

JT in my case it's worse

rogersouza-work, 2024-06-15:
JT, does not work for me either.
BTW, would you recommend someone in Spain (I live in Valencia) for recap service? 
Cheers

PapiFed, 2024-06-15:
> Let's try a shorter HS than the original. This compilation uses 3us for HS width, which is a more typical value. Let me know how it goes.
> 
> [jts18_6ae69c20.rbf.zip](https://github.com/user-attachments/files/15845819/jts18_6ae69c20.rbf.zip)

No luck on the PVM-2950.

jotego, 2024-06-15:
> JT, does not work for me either. BTW, would you recommend someone in Spain (I live in Valencia) for recap service? Cheers

Ask [here](https://www.dfcloud6.com/contacto/). Another option is [RetroArcadia](https://www.google.es/maps/place/Retro+Arcadia+Valencia/@39.4857368,-0.3789921,21z/data=!4m6!3m5!1s0xd604f9d80824661:0x37fa305d45740618!8m2!3d39.4858044!4d-0.378791!16s%2Fg%2F11j119b5yj?entry=ttu) (Tel 615237811).

rogersouza-work, 2024-06-15:
JT, RetroArcadia con Ruben. Genial!
Thanks a lot, and sorry for my misunderstood about recapping and everything.

erkkav, 2024-06-24:
I also have Sega System 18 core sync problems with my Sega Aero City and it's Nanao MS8 -monitor.  Every other Mister core works flawlessly on it. If that's valuable info I have Sega System 18 Multi (from Arcade Projects forums) too. It has Clutch Hitter as donor board. Original Clutch Hitter board and Multi afterwards has worked with the same monitor setup without problems.

Fpg-Lee, 2024-06-24:
> I also have Sega System 18 core sync problems with my Sega Aero City and it's Nanao MS8 -monitor. Every other Mister core works flawlessly on it. If that's valuable info I have Sega System 18 Multi [...]

I'm seeing quite a few owners of original S18 PCB's say they have no problem with S18 sync. .
What could be causing such incompatibility from the fpga core, that is not present with the original PCB?
Could the problem be coming from the Mister analogue board?

jotego, 2024-06-25:
As explained above, the voltage levels are different. That is not set by the core but by the external DAC.

I do not think there is nothing to do here from the core side. I am closing this issue.

Fpg-Lee, 2024-06-25:
> As explained above, the voltage levels are different. That is not set by the core but by the external DAC.
> 
> I do not think there is nothing to do here from the core side. I am closing this issue.

Hey Jose, I scrolled above, but did not see an explanation about DAC voltage levels being different from the original PCB, apologize if I missed/misread anything.
I saw your replies regarding you suggesting people get their CRT serviced, and you saying you may add further H+V sync settings to the core, what is the reality moving forward for the Mister DAC issue on S18?

Gracias.

jotego, 2024-06-25:
Here it is: https://github.com/jotego/jtcores/issues/692#issuecomment-2167927724

As I said, core-wise I cannot do anything effective. I could add more options but I think it is not going to help. I can modify the core speed, and that may lock the signal better in those screens with problems but at the cost of altering the gameplay.

MiSTer hardware vendors could add a voltage adjustment to the DAC output for this. That is rather common in supergun devices to connect arcade PCBs. This would require a new IO board or an adapter. There are probably things one can do to the signals between the MiSTer and the TV set to fix this. Most notably, changing the voltage levels, changing the line impedance and limiting the bandwidth. 

We may try to limit the bandwidth at the core output a bit, as part of the work we are doing on a different topic. That may help but I cannot know for sure.

If I had one of these screens, I would service it and replace all capacitors.

By the way, I can see a similar effect on my CRT with a completely different core. **Ghosts'n Goblins** video output does not get synchronized until the game gets to the screen full of capital H characters. After that, the TV is able to lock it in. So even the video content can have an effect on the PLL locking the synchronization signal. That clearly points to noise coupling inside the TV set. My point is that video locking is a sensitive analog signal processing and tiny things can disturb it.

jotego, 2024-06-25:
Another solution for troubled users is to use MiSTer's composite output instead of component. There is a chance that may work as the signal goes through a lot of processing that will alter the conditions for the PLL synchronization circuit.

Fpg-Lee, 2024-06-25:
> Here it is: [#692 (comment)](https://github.com/jotego/jtcores/issues/692#issuecomment-2167927724)
> 
> As I said, core-wise I cannot do anything effective. I could add more options but I think it is not going to help. I can modify the core speed, and that may lock the signal better in those screens with problems but at the cost of altering the gameplay.
> 
> MiSTer hardware vendors could add a voltage adjustment to the DAC output for this. That is rather common in supergun devices to connect arcade PCBs. This would require a new IO board or an adapter. There are probably things one can do to the signals between the MiSTer and the TV set to fix this. Most notably, changing the voltage levels, changing the line impedance and limiting the bandwidth.
> 
> We may try to limit the bandwidth at the core output a bit, as part of the work we are doing on a different topic. That may help but I cannot know for sure.
> 
> If I had one of these screens, I would service it and replace all capacitors.
> 
> By the way, I can see a similar effect on my CRT with a completely different core. **Ghosts'n Goblins** video output does not get synchronized until the game gets to the screen full of capital H characters. After that, the TV is able to lock it in. So even the video content can have an effect on the PLL locking the synchronization signal. That clearly points to noise coupling inside the TV set. My point is that video locking is a sensitive analog signal processing and tiny things can disturb it.

Sorry I did see that post before, but did not make the connection as to the difference being with the DAC.  

Is there a chance that if Directvideo mode is used via HDMI to VGA that output may play nicer with CRT's?

If not then outside of using Composite, which isn't really an option for me, then it is an unfortunate reality.

It's a shame there isn't a solution whereby the Core could increase the DAC so incrementally that the game would only be sped up by say 1HZ or so till the Screen sync's..
I'm guessing this is either not possible? or require a lot of work?

Thanks again.

jotego, 2024-06-26:
The core just do not have control over the amplitude of the HS/VS signals on the IO board.

Fpg-Lee, 2024-06-26:
> The core just do not have control over the amplitude of the HS/VS signals on the IO board.

Gotcha, now I have learned the limitation of the Core-DAC relationship regarding sync issues such as this, this has been valuable for me to learn and hopefully others.

[17 more replies]

closed 2026-11-03 by jotego