jotego/jtcores #693: Load screen stuck in loop for alternative mra files: "Street Fighter II The World Warrior" (Quicken, bootleg) AND (Quicken Pt-I, bootleg)

SensibleInt, 2024-06-03:
It writes that the RAM loads OK, then suddenly the text loses colour, and then loops before the image of the two fighters come up on the title screen.

I love this version. Was an impressive mod back in the day. Would be amazing to have it running.

Thank you for your time and incredible work.

jotego, 2024-06-03:
I was not aware that we had published MRA files for this version. Some bootleg versions were incompatible with the core/original hardware. I will check it but if it requires hardware changes to make it work (rather than changes in the MRA file), we will probably not support it. There are too many hardware variations for CPS1 bootleg boards -especially SF2- so it is a big effort and there have been no requests from Patreon supporters (especially directors) about this. Thanks for your understanding.

terminator2k2, 2024-06-03:
@SensibleInt try this mra for (Quicken, bootleg)
[Street Fighter II The World Warrior (Quicken, bootleg).zip](https://github.com/user-attachments/files/15535669/Street.Fighter.II.The.World.Warrior.Quicken.bootleg.zip)

terminator2k2, 2024-06-03:
[Street Fighter II The World Warrior (Quicken Pt-I, bootleg).zip](https://github.com/user-attachments/files/15535772/Street.Fighter.II.The.World.Warrior.Quicken.Pt-I.bootleg.zip)

SensibleInt, 2024-06-03:
You're the best!!! Thank you so much @terminator2k2 !!!!

rp-jt, 2024-06-05:
Fixed in 6b15d0f66d2e3c718804e177fcc6393113df9e4c

closed 2024-06-05 by rp-jt