jotego/jtcores #696: Pocket: add support for the Pocket Analogizer

jotego, 2024-06-04:
Links:

- [Main project repository](https://github.com/RndMnkIII/Analogizer). The project is described here.
- [Sample project](https://github.com/RndMnkIII/arcade-irem_m92_Analogizer)
- [YC module information](https://github.com/MikeS11/MiSTerFPGA_YC_Encoder)
- [YC timing table](https://docs.google.com/spreadsheets/d/12_6MyuY7f6CZYXztdT0C30SWL0nk2_MNe_Cno_D-txA/edit#gid=1960982362). We should try to avoid using it. MiSTer does not seem to need it.

RndMnkIII, 2024-06-04:
Hi, the Y/C module in Analogizer is include as standalone, not integrated in the MiSTer framework full pipeline, When instantiating it, it needs to be provided with the following parameters whose value is calculated based on the pixel clock frequency used by the core. For example for the Irem M92 core: using a base clock frequency of 40MHz needs the following parameter values:

wire [39:0] CHROMA_PHASE_INC;
wire [26:0] COLORBURST_RANGE;
wire [4:0] CHROMA_ADD;
wire [4:0] CHROMA_MULT;
wire YC_EN;
wire PALFLAG;

	parameter NTSC_REF = 3.579545;   
	parameter PAL_REF = 4.43361875;
	// Colorburst Lenth Calculation to send to Y/C Module, based on the CLK_VIDEO of the core
	localparam [6:0] COLORBURST_START = (3.7 * (CLK_VIDEO_NTSC/NTSC_REF));
	localparam [9:0] COLORBURST_NTSC_END = (9 * (CLK_VIDEO_NTSC/NTSC_REF)) + COLORBURST_START;
	localparam [9:0] COLORBURST_PAL_END = (10 * (CLK_VIDEO_PAL/PAL_REF)) + COLORBURST_START;
 
	// Parameters to be modifed
    parameter CLK_VIDEO_NTSC = 40.000000; // Must be filled E.g XX.X Hz - CLK_VIDEO
	parameter CLK_VIDEO_PAL =  40.000000; // Must be filled E.g XX.X Hz - CLK_VIDEO
	localparam [39:0] NTSC_PHASE_INC = 40'd98393783741; // ((NTSC_REF**2^40) / CLK_VIDEO_NTSC)
	localparam [39:0] PAL_PHASE_INC = 40'd121870384219; // ((PAL_REF*2^40) / CLK_VIDEO_PAL)

	// Send Parameters to Y/C Module
	assign CHROMA_PHASE_INC = NTSC_PHASE_INC; 
	assign PALFLAG = (analogizer_video_type == 4'h4) || (analogizer_video_type == 4'hC); 
    assign CHROMA_ADD = 5'd0; //yc_chroma_add_s;
    assign CHROMA_MULT = 5'd0; //yc_chroma_mult_s;
 	assign COLORBURST_RANGE = {COLORBURST_START, COLORBURST_NTSC_END, COLORBURST_PAL_END}; // Pass colorburst length

RndMnkIII, 2024-06-04:
then the parameters are passed to the Analogizer module instance:
   openFPGA_Pocket_Analogizer #(.MASTER_CLK_FREQ(40_000_000)) analogizer (
...
        //Video Y/C Encoder interface
        .PALFLAG(PALFLAG),
        .MULFLAG(1'b0),
        .CHROMA_ADD(CHROMA_ADD),
        .CHROMA_MULT(CHROMA_MULT),
        .CHROMA_PHASE_INC(CHROMA_PHASE_INC),
        .COLORBURST_RANGE(COLORBURST_RANGE),

RndMnkIII, 2024-06-04:
If you are using a well tuned value don't need to tweak nothing using CHROMA_ADD and CHROMA_MULT

RndMnkIII, 2024-06-04:
Example of interact.json file for the game In the Hunt related to Y/C settigns:
            {
                "name": "Analogizer Video Out",
                "id": 45,
                "type": "list",
                "enabled": true,
                "persist": true,
                "address": "0xA0000000",
                "defaultval": "0x0",
                "mask": "0xFFFFC3FF",
                "options": [
                    {
                      "value": "0x0",
                      "name": "RGBS"
                    },
                    {
                      "value": "0x0400",
                      "name": "RGsB"
                    },
                    {
                      "value": "0x0800",
                      "name": "YPbPr"
                    },
                    {
                      "value": "0x0C00",
                      "name": "Y/C NTSC"
                    },
                    {
                        "value": "0x1000",
                        "name": "Y/C PAL"
                    },
....

RndMnkIII, 2024-06-07:
Hi, In the following link you can see the quality obtained with SVideo using the optimized parameter values ​​for each core together with the active Y/C adapter:
https://x.com/unlovedhomie/status/1798939801485508637

RndMnkIII, 2024-06-07:
Aqui he dejado las especificaciones de Analogizer:
https://github.com/RndMnkIII/Analogizer/blob/main/specification/Especificaciones_Analogizer.md

rp-jt, 2024-06-17:
Hi,

As stated in #704, we are developing an external way to choose the settings for the Analogizer and we have chosen not to have its menu options on the OSD for the following reasons:
- As it is, it needs a lot of space that cannot be used for other options.
- It is not useful for most users, who don't posses the Analogizer.
- You have to manually choose the settings for each game.

We assume people using the Analogizer device might use mostly the same settings for CRT, so it would make sense being able to choose them once or only every once in a while and not every time you start a game.

RndMnkIII, 2024-06-17:
Of course, depending on the Analogizer menu has its drawbacks, I understand that as a general rule the Analogizer settings are not going to be changed often, in any case it would be convenient for the user to make it something easy to adjust, if possible with some external utility, and you don't have to make adjustments to configuration files manually. If you can provide it, it would be ideal.

jotego, 2024-06-17:
Yes, we'll make an easy tool for this. Hopefully updater developers will integrate this function in the updaters too.

RndMnkIII, 2024-06-17:
By the way, reviewing the code of the Y/C module used in other cores, I have seen that a version is being used that only needs two parameters, the rest are calculated internally. You would only need CHROMA_PHASE_INC and PALFLAG. Obviously for CHROMA_PHASE_INC one of two values ​​will be assigned depending on whether a PAL or NTSC signal is to be generated.

rp-jt, 2024-06-19:
I'm attaching a zip folder with two cores (simson and bubl) compiled with the options for using Analogizer. 

[release.zip](https://github.com/user-attachments/files/15903556/release.zip)

There is also a script to generate the file for option selection. You should be able to run it on a terminal and it will guide you through the options. It will create a file named "crtcfg.bin", it should be placed in release/pocket/raw/Assets/jtpatreon/common for the core to be able to read it

@RndMnkIII could you do some testing on it?

RndMnkIII, 2024-06-21:
Hello, I have only tested the video output part, as there are several details I have put it in a Google spreadsheet for you to see. Please contact me in some other alternative way.
https://docs.google.com/spreadsheets/d/1u0K5zrLRFLHMvRKUekSbgYr0MpdvuUVzaroJyC8kkTM/edit?usp=sharing

RndMnkIII, 2024-07-05:
Any news about support for Analogizer?

rp-jt, 2024-07-05:
Hello @RndMnkIII ,

We have added support for the diferent video options. About the use of SNAC, we are waiting for the adapter to arrive so we can test the implementation.

I'll leave a compilation below, so you can check the video options. There's also an updated version of the python script for option selection.

Please let us know if video is working for you as well.
[release.zip](https://github.com/user-attachments/files/16107243/release.zip)

RndMnkIII, 2024-07-05:
Hi, thanks. I'll test it. Related to SNAC, I could do the testing because already I've all the supported adapters for Analogizer.

real-amano, 2024-07-18:
@RndMnkIII why are there always two cores in the Open-fpga repos, one with analogizer support and one without it? Is there any drawback in using an analogizer enabled core without owning the analogizer? It is not explained anywhere and mostly two cores shipped, eg. neo Geo, NES, ...

jotego, 2024-07-18:
> @RndMnkIII why are there always two cores in the Open-fpga repos, one with analogizer support and one without it? Is there any drawback in using an analogizer enabled core without owning the analogizer? It is not explained anywhere just mostly two cores shipped, eg. neo Geo, NES, ...
> 
> Why not just update an existing core with analogizer as an added feature?

For JT cores, there will only be one core version. For non-JT cores, the matter is more complicated as there is not a single project with a single director. Anyway, this question falls outside the scope of this GitHub issue.

RndMnkIII, 2024-07-18:
In any case, I will respond to you but that topic is closed here since the development of the JT cores, as Jotego says, is unified and independent of the cores that I am adapting. There is no problem in maintaining the original core and the specific one for Analogizer, both can coexist together and without being independent, the Pocket menu system allows this and choosing the specific core that is going to be executed for each machine, the Analogizer cores have the core icon modified and appears as a stylized letter A. The ideal would be to unify this in each core but sometimes I have to modify or eliminate some option from the menu to make room for the Analogizer options and I have preferred to keep the two cores independently, so if you don't want to use the Analogizer features you install the normal core and so you don't see anything altered. This topic will not be discussed further here, I refer you to the Analogizer or Telegram Discord group or to my Twitter account @RndMnkIII.

rp-jt, 2024-07-19:
Hello @RndMnkIII !

You can find in the attached compilation that I added the correction for YPbPr and SNAC should be working.
Could you please test these options and give us feedback?

Thank you!
[release.zip](https://github.com/user-attachments/files/16308865/release.zip)

rp-jt, 2024-07-22:
[release.zip](https://github.com/user-attachments/files/16331619/release.zip)

In this compilation, I have added support for combining both SNAC and USB controllers with automatic assignment.
Also I have included slight changes in the SNAC modules as suggested by @RndMnkIII 

@RndMnkIII could you test it and give us some feedback?
Thank you!

closed 2024-07-26 by jotego