jotego/jtcores #700: s16/s18: enable NVRAM saving

jotego, 2024-06-08:
Saving NVRAM will save high scores. Look into adding this feature. May require moving the system RAM to BRAM.

- [ ] S16
- [ ] S16B
- [ ] S18
- [ ] shanon
- [ ] OutRun

jotego, 2024-06-09:
Preliminary work done in fea40295. 

Saving works but it tends to crash the main CPU as it block access to the RAM while saving. Restoring does not work. Not clear why.

It is better to re-start from the master branch moving the first 16kB of work RAM to BRAM in mem.yaml. Then saving and loading is taken care of by the framework.

jotego, 2024-06-10:
s16/s16b NVRAM saved support in 49052f05

jotego, 2024-06-10:
shanon/outrun NVRAM saving support in 33ae343d. OutRun does not seem to load it back correctly or at least the bookkeeping information is not stored well.

jotego, 2024-06-11:
S18 NVRAM implemented in 99cd89d0

closed 2024-06-11 by jotego

theenemy-thd, 2024-07-28:
So outrun still doesn't save hi scores? I notice in alternative roms it supports the enhanced version of outrun which does save high scores but on mister it doesn't seem to show any of the enhanced options/settings