jotego/jtcores #702: Sync Issues on PVM/Consumer CRT - TMNT, Simpsons, Splatter House

HawkeyeRetro, 2024-06-13:
I have a 6.0 I/O board and have noticed issues with sync for the TMNT, Simpsons, and Splatter House cores. I have no other sync issues with any other cores for arcades or consoles. 

My output is via component cables. I have tried these cores on two different monitors with the exact same result. One is a Toshiba A model (consumer CRT) the other is a Sony PVM 14m4u. 
[image]

[image]

https://github.com/jotego/jtcores/assets/172640919/dd917a58-331a-4ec9-8ecf-37631363f1e3

https://github.com/jotego/jtcores/assets/172640919/24b17016-d899-4065-a923-b47fcc8a3b08

jotego, 2024-06-14:
This is common for games that produce overscan images. It does not happen on our test equipment but as I said, it is not unsual. This is similar to #692. It looks like the PLL circuit in your monitors needs a bit more time to lock the horizontal synchronization signal. Hence the first few lines show that effect.

If you move the vertical image position up using the OSD menu, the PLL will get more time to lock and the top of the image will improve. But, it will push the image down. We are looking into adding more advance HS customization options in the OSD menu, but it will still not be enough for all monitors and cores. Some monitors just not lock certain frequencies in time to display the active video part of the signal.

I think replacing the capacitors on these monitors may have a chance of improving the PLL performance. Particularly, around the supply decoupling and the horizontal synchronization circuit.

closed 2024-06-14 by jotego

HawkeyeRetro, 2024-06-14:
Thank you for the reply. I have attempted to “push” the image with the OSD options but it oftentimes just makes the image worse. It is interesting that the exact same issue arises on two different monitors. Do you think this could be an issue with the 6.0 DAC? Thanks!

jotego, 2026-11-04:
There is an update planned for this week that will let you stretch in the images horizontally without distortion. Please revisit this after the update :-)