jotego/jtcores #709: Renegade: Static or incorrect blank sound?

Left-Empty, 2024-06-23:
Greetings,
I just tried after a long while to play Renegade and Nekketsu Koha Kunio-kun on the MiSTeR, and the game's sounds feature some popping static noises, or is incorrectly rendering the blank sounds? I can't tell what the issue precisely is, but it is quite evident on listening to the game.

I don't remember this being the case in the past.

Cheers!

jtmiki, 2024-06-25:
I played Renegade on MiSTer and it seemed that there were intermittent sounds plop and break.
commit: 2546510

jotego, 2024-06-26:
Using [jtopl's 34c152f](https://github.com/jotego/jtopl/commit/34c152f5ded5409a3fedcc05a56b013eec6e9924) sound is ok. Problem introduced in [jtopl's a217a0c.](https://github.com/jotego/jtopl/commit/a217a0c45c772967f8330247d52fe4dbba5fbb58).
Fixed in https://github.com/jotego/jtopl/commit/db9f6a715f7f88d2fc4bdb9a4eaa57780336faef

closed 2024-06-26 by jotego