jotego/jtcores #710: Vanguard core

jotego, 2024-06-24:
Interest from a Patreon director

real-amano, 2024-07-27:
I guess that Twin Cobra is very soon to be released by @atrac17?

thehughhefner, 2024-08-01:
> Twin Cobra seems to be released by @atrac17 soon. Just to let you know: https://x.com/_atrac17/status/1811173420832350511 (GB 2 would be my vote but I guess this is not a poll :p )

That's fine, but it shouldn't discourage Jotego from doing his version of Twin Cobra.

asturur, 2024-10-25:
As someone else already commented this should not change anything but it looks like @pcornier is working on Pandora's palace already

real-amano, 2025-01-30:
Maybe not the best place for it, but....

@gyurco , I saw that you have updated Sega 1 and Sega 2 for MiST. Would it be possible to port it to the JT framework as well, so that everyone can have it on MiSTer?

Toryalai1, 2025-01-30:
> Maybe not the best place for it, but....
> 
> [@gyurco](https://github.com/gyurco) , I saw that you have updated Sega 1 and Sega 2 for MiST. Would it be possible to port it to the JT framework as well, so that everyone can have it on MiSTer?

No need, since that core is already available in MiSTer-devel. A developer can port over the fixes that gyurco did to the repo in MiSTer-devel

real-amano, 2025-03-23:
@jotego , you added "Gun 'n Run". Which game is that? Run and gun from Konami or another one?

real-amano, 2025-08-28:
Can the jtkiwi core be revisited at some point? Chuka Taisen (=Cloud Master) is still missing and I guess that a lot of Taito knowledge was gained in the meantime.

jotego, 2026-08-15:
I prefer not to have this list as an issue as it is confusing. Hence, I am closing it.

closed 2026-08-15 by jotego