jotego/jtcores #713: CPS1 Final Fight Going to Subway blue guy alive

leitzzz, 2024-06-25:
In the first Stage when going to subway if the blue guy, is the last man stand, my character continues going to the subway, ignoring the blue guy :) i dont know if this is a bug or not. I had replicated this on mame with 2 roms that i have final fight and final fight 30th and doesnt have this problem, it fact that blue guy doesnt appear.

On the other hand this game looks so great on mister, you're doing a great job!

https://github.com/jotego/jtcores/assets/2131929/677d89f5-11e6-4a14-9dcb-b8ff6e9bb81c

jotego, 2024-06-26:
I do no see that blue guy either, but note that I test the games in a clean SD card with no previous settings of any sort. I think this is likely the effect of some DIP setting you have set in the past.

closed 2024-06-26 by jotego

leitzzz, 2024-06-26:
@jotego i tried to replicate the same in on a emulator on a raspberry pi, and happends the same, with the blue guy "jack" maybe is a bug left on the original game. 

Thanks for your response.

jotego, 2024-06-26:
How did you manage to get the blue guy on the screen?

leitzzz, 2024-06-26:
That is in a raspberry pi5 with Recalbox OS (this happends too in the Mister fpga), that guy just appear in the last building:

https://github.com/jotego/jtcores/assets/2131929/b4586b6a-2138-4995-9972-660f715c574f

Is a funny bug i think, that they left in the original game.