jotego/jtcores #721: Splatterhouse core (can't not add coins, no coin options)

AnotherPocket, 2024-07-03:
Hello. Is this Core still in beta? I am assuming that is why I can not add coins. I tried to do some research but was unable to find an answer. Thank you.

terminator2k2, 2024-07-03:
Core is still in beta phase so unless you are a patreon subscriber :

https://www.patreon.com/jotego/posts

inputs will not work 👍

ShinToK30, 2024-07-03:
The easy way to know if the jtcore is beta or public, just look  the loading or setting menu when you launch it, if its grey = public, if yellow/greenish = beta

AnotherPocket, 2024-07-03:
The loading menu is grey and the settings menu is in grey. Bionic Command looks and loads the same as all the public cores but the controls do not work after inputting coins. Splatterhouse simply does not have the ability to load coins. Previous beta cores I tried had credits that displayed over the core. Splatterhouse does not follow this pattern.

In truth, it's actually difficult to tell if a core is in beta.

I appreciate your help.

jotego, 2024-07-04:
In the MiSTer, beta cores have a yellow background in the OSD menu. That applies to JTSHOUSE (Splatter House). In the Pocket, there is no visual clue, other than the inputs not working. I should add one indeed. I guess you are a Pocket user because of your nickname. In that case, both JTSHOUSE and JTBIOCOM are in beta phase in that platform. JTBIOCOM is scheduled to become public on the next update as the only reported bug was already fixed. 

There is a list of cores in beta phase, present and past, [here](https://github.com/jotego/jtbin/wiki).

Please consider supporting the project if you find it useful.

closed 2024-07-04 by jotego

jotego, 2024-07-04:
I added a yellowish background for the OSD menu in the Pocket too. It is hard to see because of how the firmware dims it, but it is noticeable.