jotego/jtcores #727: s18: sub pixel mixing

jotego, 2024-07-12:
There is a faint yellow line around the character portrait

[image]

## Likely cause

This might be due to the vdp_sel being evaluated at the full clock rate. The vdp pixel is changing in between two pxl_cen pulses causing sub-pixel effects.

This should be visible in simulation waveforms, but maybe not in the simulation output image (as it is sampled at pxl_cen).

asturur, 2024-07-12:
just a notice that the initial screen of moonwalker that is black and white is a great example of the issue ( clearly visible ) and is immediately after boot if that helps with simulation

jotego, 2024-07-13:
> just a notice that the initial screen of moonwalker that is black and white is a great example of the issue ( clearly visible ) and is immediately after boot if that helps with simulation

Very good advice. Thank you.

Zichio, 2024-07-14:
I think these might be the same issue. I'm seeing blue outlines on both Moonwalker and Shadow Dancer ("Shadow Dancer (World).mra" and "Michael Jackson's Moonwalker (World) (FD1094-8751 317-0159).mra"). Looks like a blue outline to the left of the text. Been like this since the first release, was waiting to see if it would be fixed before reporting it.

These blue pixels are not present in MAME (tested newest version 0.267). I did not spot the issue in some direct pcb footage i watched (but that was really low quality and hard to tell 100%).

Attatching two cropped screenshots (direct screenshot capture on MiSTer).

**Moonwalker start-screen after adding a credit:**
[image]

**Shadow Dancer, intro to the first level:**
[image]

Thanks for the excellent work as always JT and Team!

gyurco, 2024-07-27:
> ## Likely cause
> 
> This might be due to the vdp_sel being evaluated at the full clock rate. The vdp pixel is changing in between two pxl_cen pulses causing sub-pixel effects.
> 
Just a note: on the schematics, while vdp_sel is combinatorial inside the PAL, the signals coming from the tilemap chip are delayed by 2 pixels (and registered with the pixel clock), due to they goes through 2 D flip-flops.

gyurco, 2024-07-31:
The bug (and the fix) also affects Super Hang-On.

jotego, 2024-08-03:
Fixed in 9807462 by @gyurco. @jtmiki mark it as completed.

closed 2024-08-03 by jotego