jotego/jtcores #729: Support for 60Hz regression

jotego, 2024-07-13:
Many systems do not output exactly 60Hz video and are troublesome for many screens (#692). Having a general fallback option to adapt the system PLL speed to produce 60Hz would fix this problem at the expense of modified gameplay speed and audio pitch.

The solution should be part of jtframe and applied to all cores automatically

- [ ] Make PLL instance configurable
- [ ] Derive 60Hz configuration parameters automatically
- [ ] Map behavior to both OSD and also a keyboard key, so it can be triggered even without access to the display

real-amano, 2024-07-13:
Shouldn't that be a per-game option?

 Usecase: "I always want un-modified gameplay speed and correct audio pitch. But game XY has a troublesome horizontal signal length for my monitor and my monitor cannot sync at all. This I have to turn it on in the game options".

MikeS11, 2024-07-13:
A better way to do it for a CRT without the sacrifice of changing the cores clocks is just to adjust the complete display window to fit the video timing better. 

I've done this for many cores in the past

[image]

See: https://github.com/MiSTer-devel/WonderSwan_MiSTer/commit/c8df028ea227bc648a01af9017cfb17b25a3dcad

MikeS11, 2024-07-13:
See this example of how it would work for shouse

[image]

jotego, 2024-07-20:
> I guess that would be a manual per-game option then? As I don't think that MISTer can know if there is monitor sync or not and thus "forced 60 hz" cannot be turned on as a fallback (eg. after x secs. without reaching sync).
> 
> Probable standard usecase: "I always want un-modified gameplay speed and correct audio pitch. But game XY has a troublesome horizontal signal length for my monitor and my monitor cannot sync at all. Thus I have to turn it on in the game options for that specific game".

That's the idea. The default is the original timing, and a 60Hz version can be turned on when needed.

jotego, 2024-09-08:
Documentation about reconfiguring the game PLL [here](https://www.intel.com/content/www/us/en/docs/programmable/683640/current/waveform-example-for-dynamic-reconfiguration.html)

closed 2025-02-14 by jotego