jotego/jtcores #734: s18: mwalk brightness too high

SumolX, 2024-07-17:
Almost all cores have similar levels except for moonwalker.  The brightness needs to be turned down on my arcade crt cabinet by a significant amount.

jotego, 2024-07-18:
How about the other s18 titles?

SumolX, 2024-07-18:
Just tested all s18 titles and they all exhibit this issue.

jotego, 2024-07-19:
This is the core output captured in simulation. The brightness is correct:

[image]

What happens in your screen is that the reference level is not being captured correctly. This is likely due to the peculiar video timing of this system. This causes synchronization problems to several users (see #692). In your case, your TV set can synchronize but with incorrect brightness and probably incorrect colors too.

So far, there is no solution for this from the core side. We may work on a tweak (#729) that will likely fix this for you at the cost of slightly different gameplay/music speed.

closed 2024-07-19 by jotego