jotego/jtcores #741: riders/simson: missing lines in objects

jotego, 2024-07-28:
Objects may miss lines under some circumstances. It can be seen in the score of _Parodius da!_ and at the end of the second stage of _Sunset Riders_.

Enemies at the top are missing lines
[image]

`CREDIT` characters are missing lines
[image]

jotego, 2024-08-18:
simulation scenes 6 and 7 in ssriders present this issue.

rp-jt, 2024-08-21:
The issue in Sunset Riders seems to be that the module painting the objects does not have enough time to finish drawing the sprites before changing the line. 
In the case of the presented scene, it appears that some more objects are drawn even being not visible, as it can be seen in this scene with only sprites:

[image]

Accelerating the process seems to solve the issue in Sunset Riders.

Right now, jtriders uses the jtsimson_obj. I am going to make an independent module for riders simplifying this (jtriders_obj) in order to solve this issue.

For Parodius, it appears that the characters presenting the issue are not in the objects layer, so a similar approach has not been useful in this case.

rp-jt, 2024-08-21:
@gyurco 
The new module for the objects will be considering only the chips 053245/053244, so it won't be compatible with Xmen anymore. Xmen will be separated to another core

gyurco, 2024-08-21:
> @gyurco The new module for the objects will be considering only the chips 053245/053244, so it won't be compatible with Xmen anymore. Xmen will be separated to another core

OK. For #759, probably it'll be even better.

jotego, 2024-08-22:
We started out with a unified module supporting the two chipsets 05324x but the code is becoming too messy. So @rp-jt will split it into two modules. One for 053244/5 and another for 053246/7. This means that Parodius support may be taken out from JTSIMSON to its own core too.

rp-jt, 2024-08-30:
Missing lines solved for ssriders in 17e9b0d28c59b28a3b2ebd7da9dc5ad6e0967dcb

Problem in Parodius not related to objects. Creating a separate issue for this

closed 2024-08-30 by rp-jt