jotego/jtcores #742: riders: wrong sprite?

jotego, 2024-07-28:
The bottom half of the pole does not look like the right sprite. Check with MAME. It might be related to the protection chip and thus we may need to wait until Furrtek finds out more about it.

[image]

There are wrong sprites moving around the saloon girls too. This happens in MAME too and it is known to be related to the protection chip
[image]

niknakniknak, 2026-09-29:
This seems fine now. Been through the whole game. no corruption anywhere.

[image]

[image]

jotego, 2026-09-29:
I don't remember any change about this...

real-amano, 2026-09-29:
That's the beauty of your modular design. 

@niknakniknak  I guess that you didn't spot the dots of bug #850 either.

niknakniknak, 2026-09-29:
You'd be right! This is what that looks like now

[image]

niknakniknak, 2026-10-27:
closing due to this not being an issue anymore

closed 2026-10-27 by niknakniknak