jotego/jtcores #743: riders: missing brightness control

jotego, 2024-07-28:
The brightness control shown on the schematics PDF page 17 is not implemented yet

jotego, 2024-07-28:
Implementation in 6a2c5050 shows bright sprites but the background is dimmed.

The output impedance of the 053535 is yet unknown but the brightness RDAC would affect black levels if implemented by translating the output voltages directly. The key here is that the TV set clamps to the dark level during blanking, so black is always black and the brightness RDAC is really limiting the maximum brightness but not the blackness.

jotego, 2024-08-09:
Added **NODIMMING** macro to disable dimming for releases until it works correctly:

``` VERILOG
`ifdef JTFRAME_RELEASE
    `define NODIMMING
`endif
```

This is always set for MiST/SiDi to allow for memory saving. It could be replaced by a simple dimming without LUT on those systems.

jotego, 2024-08-18:
simulation scene 3 shows how different dim settings make the scene look blocky and wrong

[image]

jotego, 2024-08-18:
scene 10 should look with a dimmed background and bright sprites. But everything looks dimmed instead.

[image]

rp-jt, 2024-09-18:
Fixed in eb79cbceb5394e50d8fe1955363450d1c5edec38 (#808)

closed 2024-09-18 by rp-jt