jotego/jtcores #744: Outrun graphic glitch on left side

theenemy-thd, 2024-07-28:
I have noticed an issue on the graphics on the left hand side especially on the stage with stone arches ... it happens to the left but also bad above on the top of the arch that goes the full length of the screen but again only on the left
[image]

jotego, 2024-07-29:
Thanks for reporting it. This is due to a limitation in how many points can be drawn per line without using a frame buffer for the sprites. There is no fix planned.

closed 2024-07-29 by jotego

theenemy-thd, 2024-07-29:
something that can be fixed on more powerful fpga?

jotego, 2024-07-30:
Particularly, one with more RAM chips

theenemy-thd, 2024-08-01:
You mean on board ram that has a higher access speed?