jotego/jtcores #749: Exed Exes only show colored boxes

dfilskov, 2024-08-02:
Its audio works but the video signal looks like this:

[image]

dfilskov, 2024-08-02:
Oh and as with all jt cores that rotates the video signal integer scaling is broken - it's applied _before_ the rotation instead of _after_ the rotation. 

When not rotated integer scaling works.

jotego, 2024-08-02:
Delete your Arcade folder and run update_all again. I think you're running an old version of the rbf file with a new version of the MRA file

dfilskov, 2024-08-02:
Thanks! I already did but I can double-check

dfilskov, 2024-08-02:
You were right - sorry.

I had two mra files and was trying the wrong/old one.

closed 2024-08-02 by dfilskov