jotego/jtcores #754: circus charlie core

jotego, 2024-08-06:


jotego, 2024-08-25:
executed in b23c406ba6a8aff0c786c008c49316f49bf2a282

theenemy-thd, 2024-09-01:
There seems to be an issue with graphical glitch at the top of the screen when screen flip is used in dip switches. It happens on the titlepage 
[image]
[image]
[image]
[image]

jotego, 2024-09-02:
I guess the problem is the stars going over the score? Thanks for reporting it. 

The tent going black is normal, though.

theenemy-thd, 2024-09-02:
Regarding the tent going black on thr attract screen ... at the time I re ran it non flipped and I didn't see it go black so very strange. I will test it again on friday

closed 2024-09-06 by jotego