jotego/jtcores #758: Feature request - Sinden light gun

terminator2k2, 2024-08-09:
Hi Jotego, just wandering if Sinden Lightgun support could/would be added for games like Laser Ghosts and Wheres Wally and any future light gun games? i know where's wally is not a light gun game, but it would be nice on release if you could shoot him haha.

jtmiki, 2024-08-12:
We asked to Sinden Lightgun if we could support their products.
commit: eae9a7e

terminator2k2, 2024-08-12:
Hi , Sinden sent me a message asking if you would like to get in touch with them:

 hi, if you are in contact with Jotego, I would happily send some sample Sinden lightguns for him or his team to experiment with (with no commitments), you can ask him to email contact@sindenlightgun.com with address if they are up for that, they do great work.

I have messaged Jotego.

jotego, 2025-02-23:
@rp-jt try installing the [driver](https://github.com/MrLightgun/MiSTerSindenDriver) and testing one of the cores they talk about.

jotego, 2025-03-21:
Executed in 3c2e9af58b2ae008645df0253e5990e96b57af60

closed 2025-03-21 by jotego