jotego/jtcores #762: riders: xmen. plane duplicated and misplaced for very large shrink values

jotego, 2024-08-10:
In the intro sequence, the Blackbird (X-men's plane) should appear from the distance. The zoom does not work well as it displays it duplicated for large reduction and it seems to offset it to the left and up/down. Eventually one of the two copies disappears because it seems to fall outside the screen. When the zoom shrinkage is not that aggressive, the ship appears correctly.

[image]

[image]

[image]

[image]

rp-jt, 2024-09-11:
In jt053246, vzoom/hzoom is obtained with 10 bits, but only 8 of them are considered to take the offset value from the zoffset array. In this case, vzoom/hzoom[9:8] was active, but the offset value taken was from the middle of the array, instead of an offset for large numbers of vzoom/hzoom (reduction). Something similar to what happened in #548. It can be solved in the same way

Examples in scenes 2 and 6

rp-jt, 2024-09-11:
The duplication is produced by a large value of yz_add that doesn't trigger the condition if( yz_add[16] ) inzone = 0
Can be fixed by using |yz_add[17:16] and considering larger numbers

Scene 5
[image]

closed 2024-09-12 by jotego