jotego/jtcores #765: riders: xmen review

jotego, 2024-08-11:
Feedback from toya

- [ ] Button 3 (special attack) is missing
- [ ] OSD option for dip switches/Service mode is missing
- [ ] Player shadow is covering enemies
[image]
- [ ] Magnetos magnet circle is not "circle". See image
[image]
- [ ] Sentinents mouth has sprite glitch -> maybe related to sprite zero (lime in simpsons)
[image]

- [ ] Into scene has Visual glitch on top of wolverine
[image]
- [ ] Explosions have glitches -> same as #741 
[image]

Toryalai1, 2024-08-12:
I had a look into what the button names are:
Attack
Jump
Mutant Power

[image]

jotego, 2024-10-06:
when enemies appear from the bottom, a few lines might be missing. Added **scene 11** to check this.

[image]

rp-jt, 2024-10-18:
Commit [3127b5b](https://github.com/jotego/jtcores/commit/3127b5b9fb5185192577bbe436fa218d2925252f) fixed remaining issues related to shadow priority

Since all the points mentioned above have been solved, I am going to close this issue

closed 2024-10-18 by rp-jt