jotego/jtcores #767: shouse: flip not working (Pocket)

Toryalai1, 2024-08-12:
When launching splatterhouse on the analogue pocket (fresh SD card), the game loads upside down. The Rotate Screen option in the OSD doesn't work. When I go into Service mode and toggle the flip option, it does flip, but when exiting, it goes back into upside down.  
Also, when you toggle the "Type" option in the Service menu, it also for some reason toggles the flip

Toryalai1, 2024-08-12:
Related ticket: https://github.com/jotego/jtcores/issues/691

gyurco, 2024-08-12:
You have to save the NVRAM content to save the flip option inside the service menu.
I don't know if it's there on Pocket.

Toryalai1, 2024-08-12:
There is no option to save NVRAM

jotego, 2024-08-14:
The same problem occurs in MiSTer. There must be something wrong about the DIP switch definition in the MRA file for that game.

Toryalai1, 2024-08-14:
You are correct, same also in MiSTer

irgun6, 2024-09-14:
Still not working ?

Exort, 2024-09-19:
Interestingly during the loading of the service menu boot during Ron test it does show up in the correct orientation, then flips midway through.

I can confirm that flipping in the service menu does resolve the issue but as there's no way to save it it's not useful.

There's no Flip option in the core menu, at least when this game is loaded, so there's no workaround for this issue.

jotego, 2024-09-19:
Implementing the flip hardware in commit 30b45dfb7c0ec52d97741b9a1a3f40ca9148dd4f had the side effect of revealing that horizontal games on Namco System 1 hardware are displayed through a mirror in the cabinet. The default option is to have the screen flip.

Some of them can be flipped in the service menu (like _splatter_) but in others (like _wldcourt_) the service menu fails because of a RAM error.

Not a comprehensive table:

Setname  | Service Menu   | Can be flipped
---------|----------------|--------------------
faceoff  | how to control it?    | Potentially
mmaze    | 
splatter | Works          | Yes
wldcourt | Works          | Yes

Some horizontal games do not appear flipped:

- pistoldm

Maybe the best solution will be to provide default values for the NVRAM with games preconfigured with the right flip option.

gyurco, 2024-09-19:
I also think the best would be to have a default NVRAM content then.
BTW, something's wrong with the Save NVRAM option, as after saving one, I get an EEPROM ERROR during loading. As the core supports scene dumping, I assume the NVRAM content is not (just) the EEPROM, but all memories. Or is it supposed to be correct when building a release, not a dev version?

jotego, 2024-09-19:
I just changed it today so in the release builds it will only save the 2kB for the EEPROM. I haven't tested that change, but the old one was loading the data well on the MiSTer this morning. No EEPROM errors.

gyurco, 2024-09-19:
Works fine in release mode on SiDi128.

frabico, 2024-09-20:
Splatterhouse: Flip is still not working in the new release on Analogue Pocket.

jtmiki, 2024-09-23:
I have tested Splatter House with today's compile all and Flip is still not working.
Commit: da6cb73

jotego, 2024-09-24:
The problem is that `jtframe mra` is not generating the *Saves* folder for the Pocket with the EEPROM data.

gyurco, 2024-10-04:
Meanwhile I noticed another issue: if the dump is > 2048 bytes, then the 2049th byte is also written to the EERAM, making its content useless (no saved flip setting is used). Here's a signal capture:
[image]

You can see that ioctl_addr=2048 still writes to we0_mx[0], so address 0 is overwritten in the EERAM.

gyurco, 2024-10-04:
The cause of the issue is that ioctl_wr asserted at the same time when ioctl_addr is incremented, so sel doesn't have time to de-assert. If it would be in a combinatorial block instead of a sequential, then it would work.

gyurco, 2024-10-04:
Simple diff to fix: 
'''
diff --git a/modules/jtframe/hdl/ram/jtframe_ioctl_dump.v b/modules/jtframe/hdl/ram/jtframe_ioctl_dump.v
index 5c895565..f037ab52 100644
--- a/modules/jtframe/hdl/ram/jtframe_ioctl_dump.v
+++ b/modules/jtframe/hdl/ram/jtframe_ioctl_dump.v
@@ -125,16 +125,16 @@ assign din3_mx = ioctl_ram ? {DW3==16?2:1{ioctl_dout}} : din3;
 assign din4_mx = ioctl_ram ? {DW4==16?2:1{ioctl_dout}} : din4;
 assign din5_mx = ioctl_ram ? {DW5==16?2:1{ioctl_dout}} : din5;
 
-always @(posedge clk) begin
-    sel    <= 0;
-    offset <= 0;
+always @(*) begin
+    sel    = 0;
+    offset = 0;
     if( ioctl_ram ) begin
-        if     ( ioctl_addr < OS1 && AW0!=0) begin sel[0] <= 1; offset <= 0; end
-        else if( ioctl_addr < OS2 && AW1!=0) begin sel[1] <= 1; offset <= OS1[23:0]; end
-        else if( ioctl_addr < OS3 && AW2!=0) begin sel[2] <= 1; offset <= OS2[23:0]; end
-        else if( ioctl_addr < OS4 && AW3!=0) begin sel[3] <= 1; offset <= OS3[23:0]; end
-        else if( ioctl_addr < OS5 && AW4!=0) begin sel[4] <= 1; offset <= OS4[23:0]; end
-        else if( ioctl_addr < OS6 && AW5!=0) begin sel[5] <= 1; offset <= OS5[23:0]; end
+        if     ( ioctl_addr < OS1 && AW0!=0) begin sel[0] = 1; offset = 0; end
+        else if( ioctl_addr < OS2 && AW1!=0) begin sel[1] = 1; offset = OS1[23:0]; end
+        else if( ioctl_addr < OS3 && AW2!=0) begin sel[2] = 1; offset = OS2[23:0]; end
+        else if( ioctl_addr < OS4 && AW3!=0) begin sel[3] = 1; offset = OS3[23:0]; end
+        else if( ioctl_addr < OS5 && AW4!=0) begin sel[4] = 1; offset = OS4[23:0]; end
+        else if( ioctl_addr < OS6 && AW5!=0) begin sel[5] = 1; offset = OS5[23:0]; end
     end
 end
'''

I can send a PR if desired.

gyurco, 2024-10-04:
I don't think it's a good idea to put the delay into mist's data_io. If the dump module makes delayed decision, then probably it should delay the appropriate signals itself (just reminds me to delayed cpu_dins at various places, when cpu_din is delayed, but cpu_ok is not...).
Using an ioctl_wrl inside jtframe_ioctl_dump could work, but not sure about the mister framework. If it increments address right after wr, then it'll break. Actually the block doesn't use much logic, as most of the contents are static numbers. At least it didn't broke timings in shouse.

jotego, 2024-10-04:
Ok. Let's try with the combinational approach for a while. Sometimes logic is faster than I expect.

Could you do the PR so the commit has your name?

gyurco, 2024-10-04:
PR sent!

vanfanel, 2024-11-07:
As of today, Splatter House is still inverted on the MiSTer.

jotego, 2024-11-08:
> As of today, Splatter House is still inverted on the MiSTer.

It is not inverted in 5807525 (latest version). Try deleting your saved config files.

The issue is still open for the Analogue Pocket. MiSTer is fine.

vanfanel, 2024-11-08:
I deleted every config file for the core:

`/media/fat/config# rm splatter*`

Updated to latest version via `update_all`, but it is still inverted for sure, Jose.

vanfanel, 2024-11-08:
In fact, I also overwrote  `_Arcade/cores/jtshouse.rbf` with the one here:
https://github.com/jotego/jtbin/blob/master/mister/jtshouse.rbf
...but it's still inverted.

gyurco, 2024-11-08:
I think it's not the RBF but the MRA which mostly counts. Is there a default nvram content in the MRA which you're using?

vanfanel, 2024-11-08:
> I think it's not the RBF but the MRA which mostly counts. Is there a default nvram content in the MRA which you're using?

This is the exact MRA I am using, which is the same installed by the update_all script:
https://github.com/jotego/jtbin/blob/master/mra/Splatter%20House%20(World%2C%20new%20version%20(SH3)).mra

There is an nvram section there, yes, but there's no setting related to image inversion.

vanfanel, 2024-12-02:
Still, with today's core and MRA, the game is inverted. 
Really, does someone test these arcade cores?

Toryalai1, 2024-12-02:
> Still, with today's core and MRA, the game is inverted. Really, does someone ever test these JOTEGO arcade cores?

We do test "these" jotego arcade cores. 
The game starts flipped, but can easily be fliped in the service menu.

vanfanel, 2024-12-02:
> > Still, with today's core and MRA, the game is inverted. Really, does someone ever test these JOTEGO arcade cores?
> 
> We do test "these" jotego arcade cores. The game starts flipped, but can easily be fliped in the service menu.
> 
> I wonder if it even is possible for update_all to push nvram files in update

Oh, sorry if I sounded a bit rude. Yes, it can be manually flipped on the service menu.
But I wonder how many people knows about that. I bet most users think the game is still broken...

real-amano, 2024-12-24:
@Toryalai1 Maybe the bug title can be clarified into "Screen upside down by default (must be flipped in settings)".

That hints at the solution when searching for the bug.

Toryalai1, 2025-01-06:
Solved

commit dc4548b0 (2024-09-20): shouse: default EEPROM values for horizontal games that will flip the image. Fixes #767
commit 72fbf6a0 (2024-09-24): Pocket: default NVRAM contents taken from .nvm files too. Fixes #767