jotego/jtcores #768: Arkanoid 2 - revenge of Doh spinner issues (sluggish)

fredpoulpe, 2024-08-14:
Arkanoid 2 - revenge of Doh  feels weird / sluggish with the spinner. Feels like the movement is neither smooth or realtime. I tried multiple settings but none seems good. 

note 1 : I am using a UltimateSpinner from ultimatespinner.com and Taito Egret II Mini Rollerball/Spinner Controller. Both show the same behaviour whatever the seetings used in the misterini file or on the Core. 

note 2 : I've tried OG Arkanoid, NES Arkanoid, both feel perfect.

jotego, 2024-08-14:
Try changing the sensitivity settings in the core OSD (not mister.ini). We tested that same spinner and I do not remember having issues with it.

fredpoulpe, 2024-08-14:
I tried all settings in the core and it feels like the space ship teleports instead of moving smoothly. 

Here is an example with sensitivity = high in the core. The issue is less noticeable with low sensitivity but still present and the game is less playable with low sensitivity.

https://youtu.be/oQ0J6hvEFu0?si=okk1IZEvAuOPFGfQ

jotego, 2024-08-16:
I tried it again with the current version (261a0ce) and it moves perfectly smooth. I set sensitivity to low in the core OSD menu. My mister.ini is [here](https://github.com/user-attachments/files/16638665/MiSTer.ini.zip)

I have tested with the Ultimatemister's spinner. I do not have Taito's one. I am sorry I cannot help with the problem as I do not see it at all. Maybe try asking other users in Discord too.

closed 2024-08-16 by jotego