jotego/jtcores #778: Galaga 88 flip screen not saved

spotUP, 2024-08-25:
galaga 88 (namco system 1) has an option to flip the screen in the testmenu, but the option isn’t there in the osd dipswitch settings. If i flip the screen in the test menu it isn’t saved, could this be fixed? 🙏

jotego, 2024-08-26:
You have to manually select **Save settings** in MiSTer's OSD menu. Only DIP switch changes are saved automatically by MiSTer. In this case, Galaga is saving the flip configuration to the system NVRAM. MiSTer will copy that memory when you *save settings*

closed 2024-08-26 by jotego

jotego, 2024-08-26:
You can test the files compiled for the new MiSTer framework [here](https://github.com/jotego/jtcores/actions/runs/10547910159/artifacts/1851963305), in case this is a compatibility problem with the latest mister version.

spotUP, 2024-08-26:
Huh. It works now. Sorry for the trouble!
I tried many times and was sure i saved the settings, but it seems not. Really sorry for your time spent on this.