jotego/jtcores #783: Pro Tennis World Court - Broken service menu

Retro-Nerd, 2024-08-29:
The service seems broken since the first release. I get a RAM3 Error 

011A
0000
0000
0000

After the first update for this core the image is upside down, and since we can't enter the service menu to enable the "flip screen" option... I've deleted the config files and nvram, it doesn't help here.

jotego, 2024-09-14:
Might be related to #767

jotego, 2024-09-19:
Duplicated in #767. Closing it here.

jotego, 2025-02-16:
The screen flip is fixed in 3c59f3495 by adding default contents to the NVRAM. The `RAM3 ERROR` persists.

jotego, 2025-02-16:
Tried reducing the sound RAM to AW=12, but didn't work. The RAM3 ERROR also appears randomly on MAME. As the screen flip is fixed, I'll leave the RAM3 ERROR as _not planned_ for now.

closed 2025-02-16 by jotego