jotego/jtcores #784: cps2: Add support for new hsf2 & xmvsf

Toryalai1, 2024-08-30:
With the latest release of MAME, they've added new revisions for hyper street fighter 2 and xmen vs street fighter:
Hyper Street Fighter 2 (Japan) revision 040123 as hsf2j1.
X-Men vs Street Fighter (Japan) revision 970113 as xmvsfj.

https://github.com/mamedev/mame/commit/a51d8f511f2dc765b429ea57b9a4abc820550bd0

jotego, 2024-09-15:
Done in c5c9e557b1e738a91fcbf6c97d20d58e71226728

Toryalai1, 2024-09-21:
The mentioned games are missing.

jotego, 2024-09-22:
I can see both files in JTBIN:

mra/_alternatives/_X-Men Vs. Street Fighter/X-Men Vs. Street Fighter (Japan 961023).mra
mra/_alternatives/_X-Men Vs. Street Fighter/X-Men Vs. Street Fighter (Japan 961023).mra

closed 2024-09-22 by jotego

wwark, 2024-09-24:
The issue is more those revisions appeared on mame 0.269.
But the jtcore is on 0.261 for mame regarding mra file.
The resivion name for X-Men Vs. Street Fighter is X-Men vs Street Fighter (Japan) revision 970113 as xmvsfj and not 961023

Thanks !

jotego, 2024-09-25:
We are deliberately a bit behind of the latest MAME version in order to avoid other problems. We will catch up to 0.261 eventually.