jotego/jtcores #787: paroda: Missing horizontal lines in text

rp-jt, 2024-08-30:
Issue created from #741 

CREDIT characters are missing lines
[image]

At first, we thought it was related to the objects module, but these characters are not drawn in the same layer as sprites

real-amano, 2024-10-22:
While looking at the screen shot a bit closer: is it just me or isn't the whole word affected.
It is "cred I T" with CRED affected and "I T" is not.

Hat makes the cred part different?

I even notice different spacing with I T having clearly more spacing compared to "cred".

rp-jt, 2024-11-22:
The problem seems to occur due to the cpu not having enough time to finish the instructions during blanking. 
[image]

The missing line happens when, during the first lines of the frame, the contents of this portion of the line are erased and then rewritten while the reading is happening, leading to these positions being empty when read by the scroll

The clock enable given to the cpu is 24 MHz, but effectively is being much less (20MHz). Applying a correction to give 24 MHz doesn't seem to be enough, but it helps on some scenes.

We tried changing cen_ref/cen24 entering main for 1'b1. This seems to make the problem less noticeable. This works when using JTFRAME_SDRAM96. But with the clock at 48MHz, the game does not start up.