jotego/jtcores #788: ngp color:

jtmiki, 2024-08-30:
I've tested all games of NGP Color.
These are games don't work or they works with some problems.

| Title |Problem |
|-------|-------|
| Big Bang Pro Wrestling		|blank screen
| Cool Boarders		|blank screen
| Delta Warp		|can play but can't see the title screen 
| Densha de Go! 2		|blank screen
| Dive Alert Becky's Version		|blank screen
| Dive Alert Matt's Version		|blank screen
| Fatal Fury/Garo Densetsu First Contact		|can hear music only  
| Infinity Cure		|blank screen
| Kikou Seiki Unitron		|blank screen
| The King of Fighters Battle de Paradise		|can hear music only
| The King of Fighters R-2		|can choose character only then blank screen
| The Last Blade		|can hear music only  
| Memories Off Pure		|can play but can't see the title screen  
| Metal Slug 2nd Mission		|can hear music only
| Mezase! Kanji-o		|can hear music only
| Neo Geo Gup '98 Plus 		|can play but without music
| Ogre Battle Gaiden		|can play but when a card moves to the left of the screen, can't read the description of the card
| Pac-Man		|blank screen
| Pocket Love If		|can put data but can't play
| Pocket Tennis		|blank screen
| Samurai Shodown! 2		|can play but cross-key doesn't work
| SNK Gals' Fighters		|blank screen
| SNK vs Capcom Match of the Millennium		|can hear music only

Commit: 324111

Brianvgplayer, 2024-08-30:
Pac-Man has this issue on the NGP core too. Wasn't it listed as working early on? The game has a flashing Namco logo at startup. Could that be causing issues?

Zet-sensei, 2024-08-31:
The Last Blade controls were working on version jtngpc_20240818 (MiSTer) & the game was playable with that version too during my tests session back then.

Controls aren't responding anymore on version jtngpc_20240830.

So maybe a regression to mention here for the JT team.

Zet-sensei, 2024-08-31:
Also to mention, as I don't see it on the list.

Metal Slug 1st Mission

Stuck on SNK logo.

Tested with jtngpc_20240831 (MiSTer).

Zet-sensei, 2024-08-31:
As I don't see it on the current list and this need a special note after my tests.

Some games like 'Metal Slug - 1st Mission' were redumped cleanly some time ago, without any save file attached to the ROM file.
That newer dump doesn't load on the current NGPC core (jtngpc_20240831).

You must use the older dump which contains already save init from 0x1E0000 to 0x1E00FF.
If you want the game to work on the core at the moment.

AskaLangly, 2024-09-21:
Would like to add that **Sonic Pocket Adventure** is not saving. It's playable, but it isn't able to save.
Verified by restarting the core.
Also note that by editing the game ROM to run in debug mode (edit 0x1F as FF), it refuses to save, but is more prominent with error messages. In an emulator, it does not do that.

Unedited Game CRC32: 356F0849 // jtngpc Core Version: dc4548b

rp-jt, 2025-07-09:
To summarize and being able to follow the progress of this issue:

*Blank Screen*

- [ ] Big Bang Pro Wrestling
- [ ] Cool Boarders
- [ ] Dive Alert Becky's Version
- [ ] Dive Alert Matt's Version
- [ ] Pac-Man
- [ ] The King of Fighters R-2 (During intro or after character selection: #1057 )
- [ ] Pocket Tennis color
- [ ] SNK Gals' Fighters
- [ ] SNK vs Capcom Match of the Millennium (After character selection)

*Flashing Lights*
- [ ]  Infinity Cure
- [ ] Memories Off Pure
- [ ] Pocket Love If

*Music works but unplayable*
- [ ] Kikou Seiki Unitron
- [ ] The King of Fighters Battle de Paradise
- [ ] The Last Blade
- [ ] Mezase! Kanji-o

*Save game* (#825)
- [ ] Metal Slug - 1st Mission
- [ ] Sonic Pocket Adventure 

*Controls not responding*
- [ ] Samurai Shodown! - Pocket Fighting Series
- [ ] The Last Blade

I think this is everything mentioned at the moment

rp-jt, 2025-07-09:
Revision for 09e08c610d01334021bda96d6365fc6d2f7b74a4

- I'm adding to the list Neo Dragon's Wild. It starts but it collapses on the start screen and is unplayable 

- Ogre Battle Gaiden, Last Blade and KoF Battle de Paradise seem to be working

- Kikou Seiki Unitron starts and lets you play, but some graphics are broken. For example, the borders of the chatboxes or this [map](https://youtu.be/PTz40ACJOos?t=109):

[image]

But most of the other graphics seem to be fine in this game

- Interestingly, KoF R-2 is playable in JTNGP but not in JTNGPC
- Some of the games that show a blank screen send the core to a power-off-like state where you can just load another cartridge. Similar to when you press the coin button to _"turn it off"_ before changing the game, skipping the starting settings, but after the blank screen the settings are not saved anymore and they have to be input again. For example, KoF R-2
- Other blank screens seem to collapse the core completely (Pac-Man) preventing you from turning it off or loading another cartridge

JayroJones, 2025-11-06:
SNK Gals' Fighters still boots to a blank screen, even after the Halloween update.