jotego/jtcores #789: jtbtiger is setting a wrong video mode

ojaksch, 2024-09-01:
Before this JTFriday's update Black Tiger was working fine. After the update jtbtiger tries to set a different/impossible video settings. My simple HDMI TV is showing a "mode not supported", a friend's display is saying "120Hz not supported".

jotego, 2024-09-04:
I assume that you are using a MiSTer.
I have checked the latest version of the core in several devices and I cannot reproduce your problem. Please verify your setup or try a fresh install.

ojaksch, 2024-09-04:
Forgot to mention that we're using a MiSTer. I've just re-tested the core:
- Version from 9th August is still working fine
- Version from 30th August is still "out of sync" on two setups

Very strange...

jotego, 2024-09-04:
The new version uses the latest MiSTer framework. You may need to update the MiSTer firmware.

ojaksch, 2024-09-04:
Just checked to be sure by 200% but my setup is (still) up to date, and I'm sure that my friend's too.

jotego, 2024-09-04:
I don't know then. Please try asking in Discord or user forums. 

By the way, if btiger fails for you, I would expect pretty much all CAPCOM compatible cores to fail as they share the same timings.

ojaksch, 2024-09-04:
Thanks for that hint. Tested some of your CPS1 and CPS2 core, but they're working fine.
As edited above I'll ask at misterfpga forum.

ojaksch, 2024-09-04:
Progress! Connected my Dell SE2722H, which is a MultiSync like monitor, and it displays the current jtbtiger -  at 80Hz! The older core is running at 60Hz.

paulb-nl, 2024-09-04:
For some reason the MiSTer Main is not able to set the HDMI PLL with this core. The HDMI clock will always stay at 148.5Mhz.

The resulting framerate is depending on your chosen output resolution. 720p will result in 120Hz. 1080p is 60Hz.

Only vsync_adjust 0 or 1 are affected. vsync_adjust=2 is not affected. 

This is probably caused by some timing violation since it happens with this core only.

ojaksch, 2024-09-05:
video_mode doesn't matter for me, neither 0 or 4 (my setting) changed anything except the resolution.
It was vsync_adjust indeed. It defaults to 0 and if I set this to 2 the current Black Tiger is working at 60Hz. Weird, but that shouldn't actually be the case imho.

jotego, 2024-09-06:
This should be fixed in 4972fbc1581093e745d215aae7f857f75dcd3920. I will upload the files later today. Please re-open the ticket if you see it happening again.

closed 2024-09-06 by jotego

ojaksch, 2024-09-07:
Can confirm that it's fixed now and works as with the older versions.
Thanks for listening and fixing. My money at work :heart: