jotego/jtcores #801: Vigilante Color Bug

MaskedNinjaYT, 2024-09-14:
Stage 5 Background color bug.

[image]
[image]

jotego, 2024-09-14:
This looks like a problem with ROM zip. One the bit planes seem missing. Can you check your zip file, please?

MaskedNinjaYT, 2024-09-14:
This is inside the zip:

tbp24s10_7a.ic52
vg_a-4d-.ic26
vg_a-5j-.ic37
vg_a-8h-e.ic55
vg_a-8l-a.ic57
vg_b-1b.ic1
vg_b-1d-.ic2
vg_b-1f-.ic3
vg_b-1h-.ic4
vg_b-4f-.ic34
vg_b-4j-.ic35
vg_b-4m.ic38
vg_b-6k-.ic61
vg_b-6l-.ic62
vg_b-6n-.ic63
vg_b-6p-.ic64
vg_b-8r.ic90

jotego, 2024-09-15:
I can see it too.

jotego, 2024-09-17:
- none of the values for scr2col displays the right image
- moving the CPU to clk48 did not make a difference

The problem can be reproduced with scene 2 in commit f6e38d9e

`jtsim -s 2`

Generate the sdram files by loading the game normally with `jtsim -load`. In this game, `jtutil sdram` will not get the transformation done in the game module.

jotego, 2024-09-18:
- the palette RAM content is the same as in MAME

jtmiki, 2024-09-18:
I have tested it and there are still color bugs.
[image]
Commit: 047ccd4

jotego, 2024-09-18:
The color split is visible in MAME's GFX view

[image]

jotego, 2024-09-18:
pin 13 of the palette IC was assumed to be related to the score row, but it simply marks the middle of the screen. Fixed in 34d2b4c

[image]

It works now:

[image]

commit 34d2b4ca (2024-09-18): correct background palette switch. Fixes #801
closed 2024-09-18 by jotego