jotego/jtcores #804: Add display_modes to video.json for ngp/ngpc on the Pocket

davewongillies, 2024-09-16:
Analogue have recently [added Neo Geo Pocket display modes](https://www.analogue.co/developer/docs/core-definition-files/video-json#display-mode-id-table) to firmware version 2.3. It would be nice to get these `display_modes` lists added to `video.json`.

## JTNGP `video.json`

```json
    "display_modes": [
      {
        "id": "0x61"
      },
      {
        "id": "0x10"
      },
      {
        "id": "0x20"
      },
      {
        "id": "0x30"
      },
      {
        "id": "0x40"
      },
      {
        "id": "0xE0"
      },
      {
        "id": "0xE1"
      }
    ]
```

## JTNGPC `video.json`

```
    "display_modes": [
      {
        "id": "0x62"
      },
      {
        "id": "0x63"
      },
      {
        "id": "0x10"
      },
      {
        "id": "0x20"
      },
      {
        "id": "0x30"
      },
      {
        "id": "0x40"
      },
      {
        "id": "0xE0"
      },
      {
        "id": "0xE1"
      }
    ]
```

Toryalai1, 2024-09-16:
Isn't this something that the updates can handle?

jotego, 2024-09-16:
We could leave this to the updaters, but I'm ok to adding these feature. Particularly if other cores could also use some specific modes. Do you have any suggestions of modes to add to arcade cores in general or does it work ok as it is?

jotego, 2024-09-16:
I just saw an answer to that question in this other [issue](https://github.com/jotego/jtbin/issues/399).

``` JSON
"display_modes": [
      {
        "id": "0x10"
      },
      {
         "id": "0x20"
      },
      {
         "id": "0x30"
      },
      {
         "id": "0x40"
      },
      {
         "id": "0xE0"
      },
      {
         "id": "0xE1"
      }
    ]
```

jotego, 2024-09-16:
Implemented in b89a709b
Files will be published on the next update.

closed 2024-09-16 by jotego