jotego/jtcores #825: ngp/ngpc: support for saving

jotego, 2024-09-29:
The game [Evolution](https://en.wikipedia.org/wiki/Evolution:_The_World_of_Sacred_Device) saves the progress. There is at least one user demanding this feature.

Pixelpunker, 2025-10-30:
Please don't forget this issue. It makes certain games like Sonic Pocket Adventure unsolvable.

jotego, 2025-11-03:
I assume that saving works and data is stored and restored correctly until you power off the FPGA device. Then the saved data is lost. In order to enable permanent saving, these tasks are needed

- [ ] Find out how per-game data saving is implemented on other cores for MiSTer and Analogue Pocket, both from the user and developer points of view
- [ ] Find out whether the second flash chip can be used to store data. Currently, only the first flash chip supports saving on the core
- [ ] Move the writeable flash page to an **ioctl** block in `mem.yaml`. This implies expanding the flash controller to access both SDRAM and BRAM
- [ ] Implement per-game data restoring, possibly expanding JTFRAME to support it
- [ ] It may be needed to lock writting to this page during the console boot up. There must be a mechanism to unlock it triggered by the games