jotego/jtcores #842: Xmen Konami missing echo effect.

Garagefreek, 2024-10-13:
Mame version still has problems with the sound and music in xmen but on fpga it’s sounds more accurate and close to the pcb as far as I know it’s only missing the echo effect when the attract intro arrives at the title screen, the announcement of "X-Men" has an echo effect and the same missing echo effect from Professor X's "Go Save the City".

jotego, 2025-01-18:
@Garagefreek this sound chip is important for many games, so this issue will get execute. Do not worry.

Garagefreek, 2025-01-20:
> [@Garagefreek](https://github.com/Garagefreek) this sound chip is important for many games, so this issue will get execute. Do not worry.

👍

jotego, 2026-09-24:
Furrtek's RE work is still incomplete. But I can still push the current implementation a bit further.

Garagefreek, 2026-09-25:
> Furrtek's RE work is still incomplete. But I can still push the current implementation a bit further.

When do you reckon it will be complete?

jotego, 2026-09-29:
Reverb sound effects added in 7596027caaba8f990ac40dbca38407a7a8949633. More work still in progress to close all the PCM chip details.

jotego, 2026-10-02:
Completed in 5df819afe

closed 2026-10-02 by jotego

Garagefreek, 2026-10-03:
> Completed in 5df819afe

A groundbreaking and monumental achievement - can't thank you guys enough!!