jotego/jtcores #847: Xmen Konami sentinel mouth bug.

Garagefreek, 2024-10-15:
The giant sentinel has a vertical sprite gap in the mouth..
[image]

Toryalai1, 2024-10-16:
Should've been fixed by this change:
https://github.com/jotego/jtcores/pull/831

Garagefreek, 2024-10-16:
I don't have a mister device yet I spotted this on YouTube a few days ago so I'm not sure?

Toryalai1, 2024-10-16:
What video?

Garagefreek, 2024-10-16:
Just go to YouTube and see for yourself there's a few full play through videos on xmen for the mister you'll notice the bug when you see it..

real-amano, 2024-10-17:
Well. If the YouTube video is older than the fix it is not surprising that the bug is still there in it.

closed 2024-10-17 by jotego

Garagefreek, 2024-10-17:
That's great news! thank you for clarifying much appreciated..