jotego/jtcores #857: jtkicker not showing color in y/c on MiSter

Brianvgplayer, 2024-10-29:
Both in S-Video and Composite, Kicker and some other older Konami cores like Super Basketball are showing in b/w with the MikeS adapter on MiSTer.

jotego, 2025-02-15:
@rp-jt test this one on the Pocket analogizer, which is the easiest setup we have for testing Y/C.

rp-jt, 2025-04-25:
We have tested Composite video using Pocket Analogizer on two different screens and colours are seen correctly

rp-jt, 2025-08-13:
Pictures by @TheJesusFish on Discord:

> That’s a 9.2 board using MikeS’ adapter 

[image]

[image]

jotego, 2025-08-15:
Applying this MiSTer [patch](https://github.com/MiSTer-devel/Template_MiSTer/commit/e85e20b1b9d63cf293ddcc0904d01c06b0c34e9e#diff-d219eb2a10a23096b6006901c7297ce718989d90678b46d2a1cad56d0f7aaeb9) may help these users. We should update the mister sys folder to the latest version.

TheJesusFish, 2025-08-15:
Not sure if that works with existing hardware. Might need Pork to confirm