jotego/jtcores #858: jt1942 scores are misplaced vertically and wrap around to bottom of screen.

TrickoUK, 2024-10-29:
The placement of hiscores in both 1942 and Vulugs are split where while the scores are shown at the top of the screen, the majority of the text "1UP HISCORE 2UP" is wrapped and appears at the bottom of the screen.

This is with HDMI out, vertical orientation, see screenshot for example -

[image]

All works fine still, in terms of the scores updating, its just looks off.

jtmiki, 2024-10-31:
I've checked from which core this problem started and it was from the 6th of September.
commit of 6/9/24 : dae97e2
On this date, when the game was activated, the screen went dark and only music was heard.
commit : 0af9818

jtmiki, 2024-11-04:
On the 6 of September, the Compile All was activated seven times.
The game worked fine until commit : 4972fbc, but from commit : 5803857 the screen display changed to black.

commit: 0c115ef

PR #863 (2024-11-05): Removed horizontal delay in jtframe_tilemap when flip==1
It seems that this condition became reduntant from 10c77a37a1a5a55e9569e3250bfb6e538a81312b and on.

The characters/letters layer in 1942 is drawn with this module, so, when flip==1, this layer got out of the center that the rest of the layers had, as it could be seen by the top text of 1942 and Vulgus being found at the bottom (#858 , #844 )
[image]

This change fixes #858 and #844

[image]
[image]
closed 2024-11-05 by jotego