jotego/jtcores #865: Double Dragon - Analogue Pocket (jotego.jtdd) -- Controls options does not work

AnotherPocket, 2024-11-08:
Hello. The Core Settings > Controls option for jotego.jtdd does not work. Pressing it does not respond making it impossible to remap the controls. The default set-up is a little odd for my liking or I never would have noticed this. I'm using Analogue Pocket firmware v2.3 and freshly installed jotego.jtdd as of 11/8/24 core files. Cheers!

jtmiki, 2024-11-12:
We've tested it and it worked well. 
We also tested it on different versions of Analogue Pocket and it worked well.
We will look into this a little more.
Commit : a9c8ae9

closed 2024-11-12 by jotego

AnotherPocket, 2024-11-15:
Hi. I updated Pocket OS to v2.4. I reinstalled the jtdd core (v4731644 11/9/24) and I rebuilt the ROMs (ddragon.rom & ddragonu.rom). Everything is functioning as expected. Sorry for troubling you. My previous re-installations resulted in the same experience and I assumed it was not my error. I've been told the transparent Pockets have some quirks, and assumed this was one of them. Again, my apologies. Thank you for your efforts.