jotego/jtcores #867: wc: support for grid iron

jotego, 2024-11-10:
The game starts booting but gets halted at the grid screen. Pressing 1P&2P during boot goes through the test screen, which seems ok.

The game also has an LED display, according to MAME:

>Gridiron Fight has a LED display on the control panel, to let each player
>choose the formation without letting the other know.

gyurco, 2024-11-10:
This is something about interacting with the sound CPU.
The main CPU doesn't get the expected value in the s2m sound latch, and goes to an infinite loop code path.

gyurco, 2024-11-10:
I have the fix for Gridiron, however World Cup doesn't work even with clear master. Is it expected, or am I have something wrong?

jotego, 2024-11-10:
World Cup has been running all day long. Regenerate the ROM files as I introduced an 8-byte header earlier today.

gyurco, 2024-11-10:
Yeah, ROM regeneration worked. I'll post the fix for gridiron soon.

jotego, 2024-11-11:
Thanks, gyurco. @jtmiki this issue is completed.

closed 2024-11-11 by jotego