jotego/jtcores #868: wc: dips do not work on MiST/SiDi

jotego, 2024-11-10:
b9c5d5cc5
DIPs work ok on MiSTer/Pocket. Probably related to the DIP base.

gyurco, 2024-11-10:
I don't think it's a MiST error. The address decoder doesn't look right:
```
                    3'b?0?: cab_cs  = 1;
                    2: s2m_cs  = 1;
                    4: {dip_cs,dip2_cs} = 2'b11;
                    5: {dip_cs,dip3_cs} = 2'b11;
                    // 6: wdog = 1;
                    7: {dip_cs,dip1_cs} = 2'b11;
```
The first condition overlaps with 4 and 5 (dip2_cs and dip3_cs are not even considered).

gyurco, 2024-11-10:
Actually dipsw[19:0] is OK on SiDi128. The problem is the dip2_cs and dip3_cs decoding. On the test screen, only DIP1 works, DIP2 and DIP3 are changing by the joystick (cab_cs is activated for them).

closed 2024-11-11 by jotego

jotego, 2024-11-11:
Thanks for the good hint!

commit cfd14f97 (2024-11-11): wc: cabinet address decoding corrected. Fixes #868