jotego/jtcores #884: flstory: game locks after coin+1P

jotego, 2024-12-03:
The game locks up at the beginning of the 1st stage when the broom flies away.

gyurco, 2024-12-03:
It stucks on reading D805 (MCU status), and expects 1 in bit 0 (but it only gets a zero).

gyurco, 2024-12-03:
There are two deadlocks:
- if you insert coins too fast, the sound cpu clears the input buffer flag too fast (as it already in processing the previous input), and the main cpu stuck. Probably it would even happen on the original machine, just you cannot insert coins that fast.
- at the broom scene, the MCU waits for its output consumed by the CPU, while the CPU waits for its output consumed by the MCU. I wonder if it's a timing or a logic problem.

gyurco, 2024-12-03:
Found out that pb_out[2] in the MCU could remain low (which is lrd_n), and it clears ibf even if a write to d000 happens by the CPU. I wonder if there are output enables for the GPIOs, and maybe those are also playing a role what's the effective level of these pins.

gyurco, 2024-12-03:
Hmm, the MCU sets the pb_out[2] back to '1', however the CPU already writes something to D000 while the MCU executes only 3 instructions:
```
bclr1, PORTB ; activate LWR
lda PORTA
bset1, PORTB ; deactivate LWR
```

So looks like either the MCU runs too slow or the main CPU too fast.

gyurco, 2024-12-03:
Looking at the schematics, lrd and lwr should control the flags with edge, not level.

gyurco, 2024-12-03:
Something strange with the MCU (see screenshot attached)
[image]

The instruction 'EORA $0781' should execute A^00=C0, as $781 is 00 in the ROM. However the core uses $90, which results in $C0^$90=$50, which is used later on, directing the program flow to some bogus place.

gyurco, 2024-12-03:
Actually the MCU ROM content at $081 is $90, at $781 is $00. I suspect a CPU bug with this instruction, or more generally with this addressing mode, somewhere the upper address nibble is lost.

gyurco, 2024-12-04:
Probably md_shift is missing from the ucode for this (or this group of) instruction(s)?
When fetching, 'md' became 00C8, 0007, 0081, e.g. the 07 is lost.

jotego, 2024-12-04:
The **md_shift** was definetely a bug. Quite a finding! Thank you. The microcode for the other related CPUs was find in 680x.yaml but somehow the _md_shift_ was missing in the 6805.yaml version. I have tested it both on JTBUBL and JTKUNIO, which use that MCU too. Those systems work fine, but maybe those never used that addressing mode to begin with.

Adding it makes the broomstick keep moving all the way out of the screen, which is good. But the game still locks up. 

I have also modified the MCU latch logic to follow the schematics more closely in 1d73315c2 and 594fe183a. I am not sure whether that had an effect or not.

gyurco, 2024-12-04:
After the MCU fix, it seems the tilemaps are good at least (with wrong color).

gyurco, 2024-12-04:
Something is weird with the ADCA instruction:
[image]

The code here is:
LDA $0063 ; loads 0 to A from $63
ADCA $0051,X ; X=0E, so loads 0 from 5F(51+0E), adds 0, and writes back 1!

as you can see, c=low in the capture (and in the debugger, too).
Maybe it always adds carry, even when it's not set?

gyurco, 2024-12-04:
Seems the alu adds cin_carry (from ucode) unconditionally. cx=cin_carry & cin proceeds, but still hanging...there are still other bug(s).

gyurco, 2024-12-04:
SBCA is also buggy, SBC_ALU is simply missing from the ALU, so the result is always op0. I try to add and see what happens...
(why only ADD_ALU, and not ADC_ALU, but there are SUB_ALU and SBC_ALU?)

gyurco, 2024-12-04:
Success! I'm not sure if it's the proper fix, but the game doesn't hang:
```
diff --git a/modules/jt680x/hdl/jt6805_alu.v b/modules/jt680x/hdl/jt6805_alu.v
index 42aaa54a9..bb76b1d96 100644
--- a/modules/jt680x/hdl/jt6805_alu.v
+++ b/modules/jt680x/hdl/jt6805_alu.v
@@ -40,7 +40,7 @@ assign rslt_cc = {n8,z8,c8};
 assign bsel    = {1'b0,op1[3:1]};
 
 always @* begin
-    cx = cin_carry;
+    cx = cin_carry & cin;
 
     rslt = op0;
     c8   = 0;
@@ -68,7 +68,7 @@ always @* begin
         LSL_ALU: {c8,rslt[7:0]} = {op0[7:0],1'b0};
         ROL_ALU: {c8,rslt[7:0]} = {op0[7:0],cin};
         ROR_ALU: {rslt[7:0],c8} = {cin,op0[7:0]};
-        SUB_ALU: {c8,rslt[7:0]} = {1'b0, op0[7:0]}-{1'b0,op1[7:0]}-{8'b0,cx};
+        SBC_ALU, SUB_ALU: {c8,rslt[7:0]} = {1'b0, op0[7:0]}-{1'b0,op1[7:0]}-{8'b0,cx};
         default: rslt = op0;
     endcase
```

gyurco, 2024-12-04:
I think the VRAM corruption also gone.

jotego, 2024-12-04:
That works indeed! I have the impression jt6805 is half baked. I don't see tests for it either. I wonder whether I lost some commits in a merge or something... Applying your suggestion in 7d8f7d28 fixes the locked up.

I reviewed how the carry was meant to be and found that the ASR instruction was wrong too. I fixed it in 9108301.

I think this closes this issue. I added your name in the 7d8f7d28 commit message but I wonder if you had preferred to make a PR to have you as the commit author directly.

@jtmiki this issue is completed.

closed 2024-12-04 by jotego

gyurco, 2024-12-05:
> I think this closes this issue. I added your name in the [7d8f7d2](https://github.com/jotego/jtcores/commit/7d8f7d281d3e9b54d9a9c97f5db6236c699fe79d) commit message but I wonder if you had preferred to make a PR to have you as the commit author directly.

No problem, I wasn't sure if my fix was complete (and turned out that it wasn't).