jotego/jtcores #885: flstory: no sound

jotego, 2024-12-03:
Although communication between the two CPUs seems to work, there is no sound produced

gyurco, 2024-12-05:
Spent some time (actually some hours...) with debugging. I found a bug with the flags, correcting it finally fixed the freezing with the coin insert. However still no sound...

Then I turned to mame:
- removed the AY chip from MAME - sound still there.
- removed the MSM5232 device - silence!

So this game uses the MSM5232 (almost) exclusively. Not much to do here, only implement this device.

jotego, 2024-12-06:
Blast! It is not the first time I see more sound capabilities on the hardware than what the software actually uses... Thank you for finding this out.

Could you make a PR for the flag fix?

gyurco, 2024-12-06:
Here it is:
https://github.com/jotego/jtcores/pull/888

jotego, 2024-12-06:
Thank you. I should have the MSM5232 working tomorrow.
@jtmiki, issue completed.

closed 2024-12-06 by jotego