jotego/jtcores #887: JTWC Blocked movements

jfml2020, 2024-12-05:
Tehkan when moving left or right it stops moving up and down ... I have tried everything but it remains the same ... both USB pad and Sanwa arcade. (QMTECH fpga board) .. I have discovered that if I assign movement The movements work well with buttons and not a joystick. Greetings.

jtmiki, 2024-12-09:
We have tested it on MiSTer and on QMTECH.
Buttons and joystick, both of them, work well on MiSTer but buttons don't work on QMTECH.
Commit: fe5f639

jfml2020, 2024-12-09:
It also seems that the second player's movements are reversed, right is left, etc. I don't know if it's a bug or because the game was horizontal. Greetings and thanks

jtmiki, 2024-12-10:
I have tested it with my colleague in a two-player game on MiSTer.
The second player's movements are reversed as jfml2020 said.
We also have tested it on MAME and there is a same problem.
I used yesterday's compile all. Commit: fe5f639

rp-jt, 2024-12-10:
The second player's movement is reversed because of the shape of the cabinet, where the screen was placed (literally) horizontally and both players faced each other. Therefore, player 1's "right" was player 2's "left" and so on, so they had to inverse player 2's movement so it would make sense. That is why Mame has the same behaviour

[image]

As of now, most people would not play with the screen this way, so I think it should be better adding an option for reversing player 2's movement, so it is easier to play on a conventional vertical screen setup.

real-amano, 2024-12-10:
As a two player game I would rather rotate the screen to hori? Thus Player 1 would rush towards the right side goal and Player 2 would rush towards the left goal? Or am I missing something? Thus "up" should be mapped to the right button on the pad and so on? Or am I confusing something?

jfml2020, 2024-12-10:
> As a two player game I would rather rotate the screen to hori? Thus Player 1 would rush towards the right side goal and Player 2 would rush towards the left goal? Or am I missing something? Thus "up" should be mapped to the right button on the Player 1 pad and so on? Or am I confusing something?

It was played horizontally on a ground plane as shown in the photograph above.

real-amano, 2024-12-10:
Updated my comment because I was thinking it the wrong way. Does it make sense now?

jfml2020, 2024-12-10:
> Yeah, but I guess that nobody has a cocktail setup at home with both people having the monitor between them. I updated my comment because I was thinking it the wrong way. Does it make sense now? That would be the most of logical setup for an upright cabinet mode, I think.

That's why they suggest putting a 2 player option on a normal monitor. But I guess they'll first see that everything is working like the original and then I guess they'll make the modifications.

closed 2024-12-13 by jotego

jfml2020, 2024-12-14:
Now it works with the latest modification thanks.

PR #893 (2024-12-13): Added DIP switch for selecting controller type in jtwc
Fixes #887 

Also fixes weird movement in Analogue Pocket with analog controls by adapting its standard to the one used in the framework

D-pad is set as default option, while analog controls and automatic detection have to be selected in DIP switches
commit 23e803b7 (2024-12-13): Added DIP switch for selecting type of controller. Fixes #887 and fixes movement in Analogue Pocket using analogical joysticks (jtwc)