jotego/jtcores #889: flstory: gfx problems

jotego, 2024-12-08:
- scene 2, the dragon appears partially on the screen sides, it also shows some horizontal black lines

[image]

- scene 5, part of the bubble covers Ptolemy -the main character- partially

[image]

- in a long sim `jtsim -video 800 -w 450 -inputs coin.in`, lines tend to flick, especially around the main character. Trying to capture in a scene (like scene 6) does not show the problem but a long sim shows it. It might be the main problem as the black lines in the dragon

[image]
[image]

jotego, 2024-12-08:
**Background information**

The video schematics should show how everything works. What I gather from them:

- no double line buffer, sprites to be drawn are decided during the active part of the line and drawn during blanking
- as many as 8 sprites per line are drawn. Sprites over 8 will be drawn every other frame (flicker)
- the video hardware is hard to understand, with a complex multiplexor driving the object RAM. In this RAM, there is the object data table written by the CPU, but the hardware also writes to it the object indexes to draw in the next line
- the same memory also holds column scroll values for the background, but that is not used in FLSTORY

gyurco, 2024-12-10:
I made a PR, the sprites are OK for me. I simplified it a lot by using a single line buffer.
Flipped mode is not done, but I think it's just need some offsetting in both directions.

jotego, 2024-12-11:
I think this is completed with the two PR's. Other games in the core have problems, but that would be a different issue.
Thanks @gyurco!
@jtmiki, mark as completed.

closed 2024-12-11 by jotego