jotego/jtcores #892: Using the keyboard to control Arcade games does not work

tarikalmashni, 2024-12-12:
I can define keyboard keys in the OSD just fine but when getting back into the game, non defined keys are being accepted.

Nothing happens. I can't use the coin or start button.

Tested in Super Street Fighter 2 Turbo, SSF2T New Legacy and Aliens.

asturur, 2024-12-12:
Only defaults work for some reasons in aracde. So 5 for coins, 1 for p1 start, arrows and for buttons ctrl alt shift e and something else

jotego, 2024-12-13:
Keyboard keys are hard locked to MAME's default keyboard. You can only remap controllers.

closed 2024-12-13 by jotego

tarikalmashni, 2024-12-13:
Thanks for clarification!
That's a big letdown. Is there no way around it? Other cores don't seem to suffer from this issue.

jotego, 2024-12-13:
People connecting controllers to cabinets rely on this default mapping as many of these joystick-to-USB converters look like a keyboard interface.

I think this is the first request I have got to support remapping the keyboard since I started the project 5 years ago.

tarikalmashni, 2024-12-13:
Fair enough.

I know I'm in a niche here but I've grown up playing Street Fighter on the PC with a keyboard and until now I'm still the most comfortable when playing fighting games on a keyboard.

asturur, 2024-12-13:
i grew up with the keyboard too and indeed my go to stick has a right side arrow keys to compensate that.
All the hitbox players are just negationists of the keyboard.

Jokes asides i didn't know the other cores remap keyboard either, i thought it was not working at mister level