jotego/jtcores #897: NeoGeoPocket - (RBF Mode) - Gamepad controls unresponsive after loading a game

kaloun34, 2024-12-14:
All versions after NeoGeoPocket_20240816 do not work for me, or perhaps I'm doing something wrong:

- Press the power button, which results in a white screen.
- Load the game from the OSD.

The startup screen begins to appear, but the gamepad controls do not respond.
However, everything works fine with the NeoGeoPocket_20240816 core.

jotego, 2025-01-01:
I have tested today's version, 9fae6e3, and it works correctly. Note that you must start it from the **Console** menu on MiSTer for NeoGeo Pocket. If you have a copy in **Arcade**, then the update_all script should remove it. Or just wipe out the **Arcade** folder and run update_all again, in case you have other corrupted files.

The NeoGeo Pocket _Color_ on the other hand, must be started from the **Arcade/alternatives/NeoGeo Pocket/NeoGeo Pocket color**. I know this is a bit confusing and I want to move the color version to the **Console** menu as soon as the core is made public.

closed 2025-01-01 by jotego

kaloun34, 2025-01-01:
It still doesn't work when following the ReadMe instructions:
1 -> Turn off the NGP
2 -> Load the cartridge

- The gamepad no longer responds!
- However, if I reload the core, the game launches.

https://youtube.com/shorts/XY4bvKoRwqQ