jotego/jtcores #901: [Bubble Bobble] Not possible to enter credits on the Analogue Pocket

vanfanel, 2024-12-22:
Hi there,

For some time, after updating this core, I'm not able to enter credits anymore on the Pocket, no matter what button I push.

I also tried to reset the controls, deleted all the core config files, etc

jotego, 2024-12-23:
Are you referring to the core credits?

real-amano, 2024-12-23:
Or the Highscore initials?

davewongillies, 2024-12-23:
I can confirm that the Bubble Bobble core doesn't have this problem on my Pocket.

@vanfanel try completely deleting the directories `Presets/jotego.jtbubl` & `Settings/jotego.jtbubl` then install the core again. Its probably left over files in the `Presets` causing the issue.

vanfanel, 2024-12-23:
> Are you referring to the core credits?

I mean inserting coins.

vanfanel, 2024-12-23:
> I can confirm that the Bubble Bobble core doesn't have this problem on my Pocket.
> 
> @vanfanel try completely deleting the directories `Presets/jotego.jtbubl` & `Settings/jotego.jtbubl` then install the core again. Its probably left over files in the `Presets` causing the issue.

Ah, yes, that fixed it, thanks! Closing this

closed 2024-12-23 by vanfanel