jotego/jtcores #902: Super Hang on on Analogue Pocket

Wolfyteemo, 2024-12-23:
Super Hangon was working until latest update. Now when a race starts the bike will speed up on it's own, and also after entering any direction the bike will pull constantly to the left. Tried resetting, tried reinstalling it. Tried on a fresh SD card. Still has same problem, but the game was working perfectly before the game update.

Dawbo66, 2024-12-24:
Is this docked or both ?, there's an issue docked with some controllers pulling to the left, I would put an older version on until it's fixed.

Wolfyteemo, 2024-12-24:
Not just pull to the left, it accelerates as well. This is not docked.

Dawbo66, 2024-12-24:
Yes your right also it starts and the select track screen and not the Super Hang On game screen.

jotego, 2024-12-25:
This might be related to a change done to the Pocket framework in order to have analog controls on JTWC. We have to review it.

mattyv316, 2025-01-21:
Having this problem as well, even on the latest. Going back to d29c069 (11/29) fixes the issue. Looks like the next 3b2a2fa (12/14) build is where I started having the issue.

jotego, 2025-02-14:
The commits referred above are unrelated to the control system.
This bug was fixed a while ago, probably in this [commit](https://github.com/jotego/pocket/commit/f831bb29eadd00068010da59468ed04445c19dbc) of the Pocket submodule.

closed 2025-02-14 by jotego